ThomDietrich/SenseoWifi #4: Add README

ThomDietrich, 2017-10-19:
A README with a short description, build and usage instructions needs to be added!

@guybrushix out of coincidence I saw your forking of the repository. The README is currently missing but the PCB+code are doing their job just great since one year in two machines. Do you need any help?

guybrushix, 2017-10-20:
Thank you, I first try on my own.

VinceMasuka, 2018-09-13:
I'm doing this with FHEM instead of openHAB2. Therefore some words about the MQTT functionality would be helpful. Of course it's to be read within the src files, but a litte documentation would surely help to understand readings and commands (without misinterpretation).

fredericdepuydt, 2019-01-30:
The README file currently doesn't describe the use of PlatformIO as IDE.
Took me some time before I realized the Arduino IDE wasn't the way to program the ESP8266

mboremski, 2019-07-12:
Would also like som more advice on how to bring up.
Took me some hours of research ti find hiw to get the code onto the ESP8266.
Finally found **PlatformIO** as solution.

Than it took some time to get the right libs installed and go on with compiling.
e.g. ArduinoJson has to be 5.x.x. Seems 6.x.x does some things different and wount compile.

Than tried to monitor the connected device, but default baudrate of 9600baud issn't right
So I had to add this to platformio.ini:
`monitor_speed = 115200`

With this I have been able to debug some problems with the homie/config.json

Now I am struggeling on implementing the node in openhab.
Your device uses Homie 2.x.x, but Openhab (now) wants 3.x.x thats (as far as I could figure out) not released for esp8266.
It looks like it can be added via a *.things-File manually.
Maybe you could share the neccesary steps with us?

fredericdepuydt, 2019-07-12:
@ThomDietrich Just clarifying it's build using PlatformIO, and a link to https://platformio.org/ would suffice.

@mboremski You can take a look in the development branch of homie (https://github.com/homieiot/homie-esp8266/tree/develop-v3). I'm currently still using homie 2.0 as well, and I've added the MQTT topics manually in my OpenHAB. As for the libs, when using PlatformIO in VSCode all libs are imported automatically. The baudrate can be found in 'src/SenseoWifi.cpp': `Serial.begin(115200);` (but it wouldn't do any harm if this were added in the README as well).

Kind regards,
Frederic Depuydt

mboremski, 2019-07-12:
tried to switch to dev-v3 for homie but than got a bunch of errors I cant't handle.
e.g.
```
{
	"resource": "/d:/SenseoWifi-master/src/SenseoWifi.cpp",
	"owner": "cpp",
	"severity": 8,
	"message": "no matching function for call to 'HomieNode::HomieNode(const char [8], const char [8])'",
	"startLineNumber": 22,
	"startColumn": 42,
	"endLineNumber": 22,
	"endColumn": 42
}
```
and
```
{
	"resource": "/d:/SenseoWifi-master/src/SenseoWifi.cpp",
	"owner": "_generated_diagnostic_collection_name_#0",
	"severity": 8,
	"message": "no instance of constructor \"HomieNode::HomieNode\" matches the argument list -- argument types are: (const char [8], const char [8])",
	"startLineNumber": 22,
	"startColumn": 22,
	"endLineNumber": 22,
	"endColumn": 22
}
```
fogot to say:
espressif8266 > 2.0.4 also generates some problems.
Needed to change platform in platformio.ini to:
`platform = espressif8266@2.0.4`

would you share your files with us to help bring it up?
looking for things-file, map-file and sitemap-file(sniped)

mboremski, 2019-07-13:
Have a look at #19 to see some config-example for OH  2.4.0 ;-)

ThomDietrich, 2020-02-08:
Hey guys, I've updated the README to include flashing instructions and other details. Please open a new issue for whatever you run into

closed 2020-02-08 by ThomDietrich