ThomDietrich/SenseoWifi #7: test environment starts to beep immediately after flash

VinceMasuka, 2018-05-14:
Hi

I just rebuild your project. Right after flashing the buzzer starts to beep quiet annoying. Of corse I want to test the board before I start to integrate it into the Senseo. That's why (I guess) I should connect 5V and GND only to start WiFi. Unfortunately there is this noisy beep again (and no WiFi). Perhaps it's just my confused, deaf ears for the moment, but right now I'm kind of clueless what this thing wants to tell me...

Suggestions? Anyone?

Thanks!

Regards
Vince

ThomDietrich, 2018-05-15:
Hey!
Well that shouldn't happen but it is along time since I last flashed without a connection to the machine. Is the beeping continuous?

My suggestion would be to 1) check the serial console and/or the MQTT broker to learn about the state of the program. Besides that, 2) you could disable the buzzer in the initial configuration custom settings.

VinceMasuka, 2018-07-01:
Sorry... I was like unable to work with this project for the past several weeks...

Yes, it beeps continously. I think I leave this issue at "continous improvement process". Means that I unsoldered the beeper for silent's sake and try to get the rest running and build in the Senseo until I get back to this. Perhaps it's like you said and only an issue if it's not build in.

ThomDietrich, 2018-07-02:
I'd suggest to first connect everything and then check the serial console for debugging/log ouptut. You'll learn about the actions of the program. A constantly beeping of the beeper tells me that you probably did not yet flash my source code onto the chip. There is no continuous beeping in my code...

closed 2020-02-08 by ThomDietrich