ThomDietrich/SenseoWifi #9: openHAB - dozens of items

VinceMasuka, 2018-07-01:
Hi

Do you mind to please...
- share your dozens of openHAB items
- explain a little how you did the binding and
- of course how you did the Amazon Alexa Echo implementation?

Thanks in advance!

Regards
Vince

VinceMasuka, 2018-07-02:
Now that is awesome, too! Thanks a lot for the templates and the hint with the openHAB cloud addin.

With binding I meant the bindings within openHAB2. As there is openHAB2, but there still just is this 1.x implementation of the MQTT binding. Also there is this beta 2.x binding, but is able to mess up your whole openHAB2...

Let me guess:
You took a mqtt broker like mosquitto and modified the mqtt.cfg to communicate with each other. Than you handled the items, grouped em and somehow got it to work?

ThomDietrich, 2018-07-03:
Go with openHAB and the 1.x binding.

The new binding will come eventually and this will be awesome because it will be able to auto-detect all items for my firmware.

Anyhow, using the 1.x binding is not complicated. In fact I wrote a tutorial about it: https://community.openhab.org/t/mqtt-binding-v1-11-getting-started-101/33958

closed 2018-12-17 by VinceMasuka