ThomDietrich/SenseoWifi #10: Error in fz-File (fritzing 0.9.3b on Windows 10)

evildrummer, 2018-07-16:
[image]

ThomDietrich, 2018-07-16:
Nope sorry must be an issue with your system or download. Just tested with a fresh Fritzing and file download myself.

evildrummer, 2018-07-16:
Hm,

i tested it on Win7 and 10 (x86 and x64 Version). :-/ 

Can you upload a fzz or gerber version of the File? Then i dont need Fritzing anymore.
Fzz is the supported type when i want to upload it on their Website (aisler.net). 

Thanks

ThomDietrich, 2018-07-16:
[SenseoWifi-PCB_GERBER.zip](https://github.com/ThomDietrich/SenseoWifi/files/2197631/SenseoWifi-PCB_GERBER.zip)

This is the file I last used to produce PCBs I'm currently using. Be aware that I realized later on that two connectors are mission in this version https://github.com/ThomDietrich/SenseoWifi/issues/8#issuecomment-389109496

ThomDietrich, 2018-07-16:
You should try once again. Your error sounds like you did something wrong when downloading the file...

evildrummer, 2018-07-16:
Well. Finally i was able to open it :)

but!..... i´m not sure if my added pins are well placed :) 
[image]

If you´re giving me an ok i´ll do the pull request. But first i have to check out what that means :) I´m new on github.

ThomDietrich, 2018-07-16:
Looks fine to me. I myself was hoping to not extend the size of the PCB... you know what, give me a sec.

ThomDietrich, 2018-07-16:
[image]

ThomDietrich, 2018-07-16:
[SenseoWifi_PCB_v1.7_Extended_Gerber.zip](https://github.com/ThomDietrich/SenseoWifi/files/2198093/SenseoWifi_PCB_v1.7_Extended_Gerber.zip)

ThomDietrich, 2018-07-16:
The new connectors on the bottom left are for the Reset button and for VCC from the step down converter. Enjoy! Let me know if you need more help!

mboremski, 2019-07-20:
@ThomDietrich might be closed.. not? :-)

ThomDietrich, 2019-07-22:
Sounds like it. Thanks ;)

closed 2019-07-22 by ThomDietrich