ThomDietrich/SenseoWifi #12: (Improvement) Replace Step Down AC-DC Converter (220V to 5V) by Hi-Link-ac-dc

diogopms, 2018-09-10:
Note: This is not a real issue, but an improvement to the PCB.

What do you think, about changing to:
https://pt.aliexpress.com/item/Hi-Link-ac-dc-power-module-5v-3w-220v-isolated-ac-dc-converter/32360232521.html

VinceMasuka, 2018-09-13:
Works flawlessly. I took the same on my setup.
[http://www.hlktech.net/product_detail.php?ProId=54](url)

ThomDietrich, 2018-11-21:
Looks good indeed. Why would you recommend this one over the one I've linked?

VinceMasuka, 2018-12-17:
Perhaps because it comes in a enclosure?
[image]
(inside it's pretty much the same, I guess)

Actually that might be preferable of sorts, as there is water around and you might would like the idea of some extra "shielding"...

ThomDietrich, 2018-12-17:
Fine by me. Please create a PR against the README.md as an alternative product in the components table.

Ditoxius, 2019-01-17:
Hi, i have another question regarding the supply voltage.

The senseo µC is powered the whole time with 5V via a capacitive divider and zener diode. Have you tried using its supply voltage to power the WeMos module instead of installing a separat step-down converter?

Thank you for your project....i hope to make my coffee maker smart, soon :D

ThomDietrich, 2019-01-18:
Yes. I tried and I did a few thought experiments on the topic with colleagues. The problem is (please correct me if you disagree) that such a circuit has a narrow operating point and experiments showed that voltage drops as soon as more than a few milliamps are consumed.

Another problem is the missing divide of electrical potentials between the Senseo supply and the USB connection (think optocouplers).

I very much prefer the decoupled solution for its simplicity and the flexibility in build in troubleshooting. Lastly and maybe most importantly: The solution comes as safe as it can get (I hope :) ).

closed 2019-01-18 by ThomDietrich