ThomDietrich/SenseoWifi #13: current on the buttons

VinceMasuka, 2018-09-13:
Hi...

I tried to get this project finally done and everything is now build in the Senseo:

- MQTT nicely reports the readings

- MQTT readings instantly update themselves when you send a MQTT command or

- press a button on the Senseo

But:

- a MQTT send "power" to the board reports that Senseo is powered, but it won't turn on (harware button works of course)

- strangely all buttons on the board (TP, T1, T2) allways have current of exactly 5.08 Volts (no matter if MQTT says they are on or not) (I expected something like 0 Volts when it's turned off)

By any chance you can't push me in the right direction what to check first?

Thanks!

Regards
Vince

ThomDietrich, 2018-11-21:
Hey Vince!
Sorry for the late reply. Did you solve the issue?

ThomDietrich, 2018-12-14:
Hey!
Just to summarize:

1. You receive MQTT messages from the board and therefore are certain that the firmware flash worked.
2. More precisely, you receive "ON" via `.../power` and "SENSEO_HEATING" via `.../opState` if you press the power button manually?
3. You publish a message with payload "ON" to the topic `.../power/set` and "MQTT reports back" - what do you mean by that? As you can see [here](https://github.com/ThomDietrich/SenseoWifi/blob/master/src/SenseoWifi.cpp#L45) there is no optimistic response. The response should be what's described under (2.). For testing please send an invalid payload and check for the "power: malformed message content." debugging message.

To rule out hardware issues you could modify the source code to trigger a power press, e.g., when the firmware has booted.

ThomDietrich, 2018-12-14:
FYI:
Try to get it working with MQTT first. That is best for troubleshooting.
For later use you might be happy to find my openHAB configuration here: https://github.com/ThomDietrich/openhab-config/blob/master/items/senseoWifi.items
Don't torture yourself with FHEM 😅

VinceMasuka, 2018-12-16:
1. I really like to belive that.
2. Yes, that works as expected. If I manually push the button physically on the Senseo, my mosquitto_sub tells me "opState SENSEO_HEATING", "power ON", "outOfWater false", "brew false" and "opState SENSEO_READY".
3. Oh, I just meant if I publish something, that the mosquitto subscription immediately tells me so (as expected). So when I send a invalid content like
"mosquitto_pub -t homie/Senseo/machine/power/set -m toggle"
I receive:
"debug power: malformed message content. Allowed: [ON,OFF]"

That brings us to the assumed valid version:
mosquitto_pub -t homie/Senseo/machine/power/set -m ON
...but "opState" stays on "SENSEO_OFF".

ThomDietrich, 2018-12-16:
So far so good. Now I'm afraid you need to modify the source code to figure out where the issue lies exactly.

VinceMasuka, 2018-12-17:
OK, for the sake of progress, I'll have to do so...

So this suggestion implies that you think this is not hardware, but software related?

ThomDietrich, 2018-12-17:
To the contrary. As the firmware seems to work and interact with your machine (it's able to detect "power ON") this is most probably an issue with your hardware setup. You need to follow along the route, make sure the optocoupler is swithced and check why it doesn't trigger a Senseo button press. One thing you could do: Increase the milliseconds of virtual button press, then measure the optocoupler on output side. Whatever you do please be careful with high voltage.

VinceMasuka, 2018-12-18:
Nice hint. I modified the virtual button press to 1000 an voila: Senseo is powering up.

Thanks!

ThomDietrich, 2018-12-19:
Wonderful. Would you invest a few extra minutes to figure out which lowest value would work for you? I'd like to update the source code here accordingly but one full second is quite a lot. How about e.g. 150 or 200?

VinceMasuka, 2018-12-19:
As to your pictures and decriptions, having your project as example for mine, I directly added an external micro usb access to my 7820 Senseo, too. This makes flashing pretty easy now and I'll do some more flashes (see my open TCRT issue). With this upcomming flashing, I'll alter the virtual press button and report back within this thread.

ThomDietrich, 2018-12-19:
Thanks! Your help will make a rebuild easier for the next guy.

I need to update documentation. I know that it does not at all cover all aspects of the project.
Regarding OTA I can tell you that you need to follow the OTA mechanism of homie-esp8266:

* http://marvinroger.github.io/homie-esp8266/docs/stable/others/ota-configuration-updates/

If the script doesn't work for you check out this bash script: https://github.com/marvinroger/homie-esp8266/issues/308

VinceMasuka, 2018-12-21:
I don't know what went wrong in first place, but it now seems that my setup accepts much lower ms for the virtual button press than before (I'm starting to suspect a faulty copper wire in there...).

Anyhow it looks like that
    _static const int pressDuration = 36;_
is the lowest working value with the Seseno 7820.

mboremski, 2019-07-15:
Not sure if this still is open..
I also had some issues to get the buttons "pressed" via OP.
For me 110ms have been the lowest value it was working with.
I would recommend to change it to:

```cpp
// Duration of button "press" (in milliseconds)
static const int pressDuration = 150;
```

ThomDietrich, 2019-07-19:
It is probably safe to assume that this change will not have any side effects. I would be most curious to know why this is an issue for you guys. Do you own an older/newer Senseo model or did you use different optocouplers?

mboremski, 2019-07-20:
From my knowledge of µC coding it **might** be possible that Senseo changed the time a button needs to be pressed. Maybe for better antibeat performance.
I think the 100ms has been a bit to tight to fir for all devices.

Did you once try to shorten your `pressDuration` to find the minimum pressing-time for your Senseo?
Could be that 90ms already is to short. In this case the 100ms are to tight to adress also manufacturing-tolerances.

ThomDietrich, 2019-07-22:
I probably did. My initial tests to understand the behavior of the machine were pretty thorough. Can't remember though and your 90ms argument could just as well be true. 150 is a worthy improvement 👍

closed 2019-07-22 by ThomDietrich