ThomDietrich/SenseoWifi #14: warning while compiling - SENSEO_unknown not handled...

VinceMasuka, 2018-09-16:
Hi

Found something odd while compiling the code:

`Compiling .pioenvs\d1_mini\libcd1\ESP8266WiFi\ESP8266WiFi.cpp.o
src\SenseoWifi.cpp: In function 'void senseoStateExitAction()':
src\SenseoWifi.cpp:126:10: warning: enumeration value 'SENSEO_unknown' not handled in switch [-Wswitch]
switch (mySenseoSM.getStatePrev()) {
^
src\SenseoWifi.cpp: In function 'void senseoStateEntryAction()':

src\SenseoWifi.cpp:198:10: warning: enumeration value 'SENSEO_unknown' not handled in switch [-Wswitch]
switch (mySenseoSM.getState()) {
^
In file included from .piolibdeps\Homie\src/Homie.h:4:0,
from src\SenseoWifi.cpp:7:
src\SenseoWifi.cpp: In function 'void setup()':
.piolibdeps\Homie\src/Homie.hpp:27:214: warning: statement has no effect [-Wunused-value]
#define Homie_setFirmware(name, version) const char* __FLAGGED_FW_NAME = "\xbf\x84\xe4\x13\x54" name "\x93\x44\x6b\xa7\x75"; const 
char* __FLAGGED_FW_VERSION = "\x6a\x3f\x3e\x0e\xe1" version "\xb0\x30\x48\xd4\x1a"; Homie.__setFirmware(__FLAGGED_FW_NAME, __FLAGGE
D_FW_VERSION);
^
src\SenseoWifi.cpp:352:3: note: in expansion of macro 'Homie_setFirmware'
Homie_setFirmware("senseo-wifi-wemos", senseoWifiFWVersion);`

Some warnings are not an error, but strange enough...

Regards
Vince

ThomDietrich, 2018-11-21:
Not critical but definitely worth correcting

ThomDietrich, 2018-12-17:
I'll correct these warnings soon. Thanks for reporting!

ThomDietrich, 2020-02-08:
Done

closed 2020-02-08 by ThomDietrich