ThomDietrich/SenseoWifi #15: publishes to MQTT but does not receive

vogler, 2018-11-19:
I'm using home-assistant's embedded MQTT broker. The esp8266 publishes topics like opState, but doesn't receive any commands. How can I debug this?

vogler, 2018-11-19:
Ok, just found out it's actually `devices/senseo-wifi/machine/power/set`. Works 😄 
Maybe a sentence in the readme on how to test it would be a good idea.

ThomDietrich, 2018-11-21:
The README does not explicitly state that but the device follows the Homie convention. Check it out for all details! ;)

I'll need to update the README with a lot of things. Anything else you found? Does your machine work? Btw if you are located in Germany I could send you one of my spare PCBs.

ThomDietrich, 2018-11-22:
Hey!
You can leave USB connected for debugging, that is the main reason for the optocouplers.

What concerns me is the picture of your machine. It seems to be an older model. Could you tell me the exact type number?
Nevertheless, as the PCB just observes the LED and digitally pushed the buttons I believe it should still work. I would suggest to connect USB, then go through a bit of testing. You should see *very precise* timings of +-1ms. See: https://github.com/ThomDietrich/SenseoWifi/blob/master/src/constants.h#L12 for the times.

> but now I also get values like 6 and 285

Please check again. If those are your readings there is probably something wrong with your PCB or power supply.

If your Senseo PCB is older, this could also be an issue with a less stable LED power supply. Be careful while playing around with this part of the problem!!

Please keep me in the loop!

Also: Any addition to the readme, the source code or the PCB is highly welcome!!

vogler, 2018-11-22:
It's a Philips HD 7801/62.
Everything worked fine for the first brew. I found out that now `ledChangedHandler` seems to fire whenever `cupAvailable` changes. When I remove the TCRT5000 it doesn't change states anymore.

I use the following times:
~~~
pulseDurLedSlow = 1000;                  
pulseDurLedFast = 80;                    
pulseDurTolerance = 30;                  
pulseContThreshold = 2 * pulseDurLedSlow;

HeatingTime = 70; // cold start -> LED blinks -> LED on
HeatingTimeTol = 10;
Brew1CupSeconds = 34; // warmed up: LED on -> LED blinks -> LED on
Brew2CupSeconds = 70; // "
BrewHeat1CupSeconds = 64; // HeatingTime+Brew1CupSeconds
BrewHeat2CupSeconds = 140; // HeatingTime+Brew2CupSeconds
~~~
It wasn't exactly clear to me what the difference is, are the comments right?

ThomDietrich, 2018-11-27:
Before working with the recipe feature I'd recommend to find out if the rest does what it is supposed to. If the TCRT5000 seems to be part of the issue could you please check if the problem is on PCB or code side?

Regarding the times: Are you sure your changes are correct? Does your machine really have a fast blinking period of 80?

The comments are not clear indeed. Please create a PR ;) See here: https://github.com/ThomDietrich/SenseoWifi/blob/master/src/SenseoWifi.cpp#L139 BrewHeat is the normal cycle from pressing the brew button till the machine is finished and the LED stops blinking. The brew times represent the first part of those if the machine switches to OFF or NOWATER after brewing.

evildrummer, 2018-12-29:
Ich hätte gerne das übrige pcb abgekauft :)

closed 2020-02-08 by ThomDietrich