ThomDietrich/SenseoWifi #16: TCRT5000 switched

VinceMasuka, 2018-12-18:
Hi there

I wired everything with a TCRT5000 connected to the "main" board. Funny thing is, that without cup both leds on the TCRT are on and the MQTT subscription says
_(...)/cupAvailable true_

Now when I put a cup in place, one of the LED on the A0 side turns of and MQTT reports
_(...)/cupAvailable false_

Now shouldn't be that the other way around?

ThomDietrich, 2018-12-19:
Most certainly 😄 My best guess is that you've bought a module that behaves differently than mine ([compare here](https://www.ebay.de/itm/TCRT5000-Reflektierende-IR-Lichtschranke-Optisch-Sensor-Modul-Arduino-Raspberry/252795025375?hash=item3adbc1f7df:g:AUkAAOSwmUBb3x-Z:rk:2:pf:0)).

I believe you just need to remove the "!" from those two lines: https://github.com/ThomDietrich/SenseoWifi/blob/master/src/Cup.cpp#L12-L16

ThomDietrich, 2018-12-19:
Btw you might want to activate this to test everything: https://github.com/ThomDietrich/SenseoWifi/blob/master/src/SenseoWifi.cpp#L340-L346

VinceMasuka, 2018-12-21:
Awesome: "We have a bingo here!"

Somehow I just thought a TCRT5000 would be a TCRT5000, but how could I... Well, this one is mine:
[https://www.amazon.com/WINGONEER-TCRT5000-Infrared-Reflectance-Avoidance/dp/B01DCZA6RG](url)
(and it likes to have the exclamation marks out there)

Thanks a lot!

closed 2018-12-21 by VinceMasuka