ThomDietrich/SenseoWifi #18: Toggle cup detection sensor

fredericdepuydt, 2019-01-30:
Hi,

A recently started creating this project myself, and was wondering how to minimize the power consumption. I'm suspecting the cup detection will be lit 24/7, but the actual detection only needs milliseconds to determine if there's a cup or not.

Wouldn't it be interesting to have a digital output pin controlling the input voltage of the TCRT5000 module?

Kind regards,
Frederic

ThomDietrich, 2019-07-15:
Hey Frederic,
did you measure the current on the module? Even with the LED it's probably just a few milliamps and would cost you literally cents :) To be fair I never measured the consumption of the whole setup. Would be interesting to know which difference the added components make relative to the normal standby consumption of the Senseo.
Regarding the LEDs: Did you consider disconnecting it?

fredericdepuydt, 2019-07-15:
> Hey Frederic,
> did you measure the current on the module? Even with the LED it's probably just a few milliamps and would cost you literally cents :) To be fair I never measured the consumption of the [...]

I haven't measured it yet, I'll do some measurements when my project is finished ;) I'll be using an IR led with a transistor and an IR sensor with a transistor. This allows me to first measure the ambient light, and afterwards compare the value when the IR led is turned on.

This measurement would only take several milli seconds, and during idle times both transistors would be turned off. I'll share the schematics and code when it's finished.

ThomDietrich, 2019-07-16:
> I haven't measured it yet, I'll do some measurements when my project is finished ;) I'll be using an IR led with a transistor and an IR sensor with a transistor. This allows me to first measure the ambient light, and afterwards compare the value when the IR led is turned on.

The TCRT5000 also has an analog out, giving you a comparable measurement. Then you can simply check for differences and thereby identify the cup. I won't stop you, not sure if it's worth it :) If you want to improve something it would be interesting to learn about a solution that is more easier to install and maintain. Cutting the hole for the two LEDs and keeping them clean is not easy. I would love a sensor that is e.g. housed in one round shape and flush with the senseo surface...

In any case: Yes definitely share your findings! :)

mboremski, 2019-07-20:
I have been corius and measured the current of the TCRT5000:
Without cup: 25.7mA
With cup: 28.9mA

So the Sensor is taking approx. 145mW continuously.
This leads to 1.27kWh per year.
That is in germany around 0,4€/year. ;-)

ThomDietrich, 2019-07-22:
Thanks @mboremski !!

> The whole PCB is taking around 110mA (including TCRT5000)

would you like to add this to our README? I can see others raising similar questions.

ThomDietrich, 2020-02-08:
Even if measurements by @mboremski are not correct, I guess it is safe to assume that this is not a real issue. I've added a note to the README. https://github.com/ThomDietrich/SenseoWifi/commit/fc12187822dca027e2fcc0dc260cb7dc9552dedc

closed 2020-02-08 by ThomDietrich