ThomDietrich/SenseoWifi #19: Configuration example for OH 2.4.0

mboremski, 2019-07-13:
Hey,
after some hours of configuration I'd like to share my config-files for your convinience:

mqtt.things
```
Bridge mqtt:systemBroker:LocalBroker [ host="your-broker-ip", secure=false ]
{
    Thing mqtt:topic:SenseoWiFi {
    Channels:
        Type string : Debug           "Debug"               [stateTopic="devices/senseo-wifi-home/machine/debug"]
        Type string : OpState         "Zustand"             [stateTopic="devices/senseo-wifi-home/machine/opState"]
        Type switch : OnOff           "OnOff"               [stateTopic="devices/senseo-wifi-home/machine/power", commandTopic="devices/senseo-wifi-home/machine/power/set", on="ON", off="OFF"]
        Type string : Brew            "Brew"                [stateTopic="devices/senseo-wifi-home/machine/brew",commandTopic="devices/senseo-wifi-home/machine/brew/set"]
        Type number : BrewedSize      "BrewedSize"          [stateTopic="devices/senseo-wifi-home/machine/brewedSize"]
        Type string : OutOfWater      "OutOfWater"          [stateTopic="devices/senseo-wifi-home/machine/outOfWater"]
        Type string : Recipe          "Recipe"              [stateTopic="devices/senseo-wifi-home/machine/recipe",commandTopic="devices/senseo-wifi-home/machine/recipe/set"]
        Type string : CupAvailable    "Tasse vorh."         [stateTopic="devices/senseo-wifi-home/machine/cupAvailable"]
        Type string : CupFull         "Tasse voll"          [stateTopic="devices/senseo-wifi-home/machine/cupFull"]
        Type string : Online          "Online"              [stateTopic="devices/senseo-wifi-home/$online"]
    }
}
```
SenseoWifi.items:
```
String KU_Senseo_Debug         "Debug [%s]"                          {channel="mqtt:topic:SenseoWiFi:Debug", expire="10s"}
String KU_Senseo_OpState       "Zustand [MAP(senseo-wifi.map):%s]"   {channel="mqtt:topic:SenseoWiFi:OpState"}
Switch KU_Senseo_OnOff         "OnOff"                               {channel="mqtt:topic:SenseoWiFi:OnOff"}
String KU_Senseo_Brew          "Brew"                                {channel="mqtt:topic:SenseoWiFi:Brew"}
Number KU_Senseo_BrewedSize    "BrewedSize"                          {channel="mqtt:topic:SenseoWiFi:BrewedSize"}
String KU_Senseo_OutOfWater    "OutOfWater"                          {channel="mqtt:topic:SenseoWiFi:OutOfWater" [profile="transform:MAP", function="senseo-wifi.map"]}
String KU_Senseo_Recipe        "Recipe"                              {channel="mqtt:topic:SenseoWiFi:Recipe"}
String KU_Senseo_CupAvailable  "Tasse vorh."                         {channel="mqtt:topic:SenseoWiFi:CupAvailable" [profile="transform:MAP", function="senseo-wifi.map"]}
String KU_Senseo_CupFull       "Tasse voll"                          {channel="mqtt:topic:SenseoWiFi:CupFull" [profile="transform:MAP", function="senseo-wifi.map"]}
String KU_Senseo_Online        "Online"                              {channel="mqtt:topic:SenseoWiFi:Online"}
```
SenseoWifi.sitemap
```
sitemap SenseoWifi label="Senseo-Tests"
{
    Frame label="Status" {
        //Text item=KU_Senseo_Unreach label="Fehler [Maschine nicht verfügbar!]" visibility=[KU_Senseo_Online == "false"] valuecolor=["true"="red"]
        Text item=KU_Senseo_OpState label="Aktueller Zustand"
        Text item=KU_Senseo_OutOfWater label="Wassertank [Leer]" visibility=[KU_Senseo_OpState == SENSEO_NOWATER] valuecolor=[ON="maroon"] icon="water"
        Switch item=KU_Senseo_CupAvailable label="Tasse vorhanden"
        Switch item=KU_Senseo_CupFull label="Tasse voll" visibility=[KU_Senseo_CupAvailable == ON]
    }
    Frame label="Steuerung" {
        Switch item=KU_Senseo_OnOff label="Ein-/Ausschalten"
        Switch item=KU_Senseo_Brew label="Kaffee Kochen" mappings=[1cup="☕", 2cup="☕☕"]
    }
}
```

Please be aware:
* Conversion of string-states "true"&"false" is a bit rude i think.
* There also have only been some basic testing because my hardware is not in place yet.
* Rules have not been implemented

Suggestions are welcome.

@ThomDietrich could you maybe comment or even add to readme?

fredericdepuydt, 2019-07-15:
I've also added the Homie2.0 default stats

`Type string : version        "Version"          [ stateTopic="devices/senseo/$homie"]
Type string : name           "Name"             [ stateTopic="devices/senseo/$name"]    
Type string : ip             "IP-address"       [ stateTopic="devices/senseo/$localip"]
Type string : mac            "MAC-address"      [ stateTopic="devices/senseo/$mac"]
Type string : fw_name        "Firmware Name"    [ stateTopic="devices/senseo/$fw/name"]
Type string : fw_version     "Firmware Version" [ stateTopic="devices/senseo/$fw/version"]
Type string : nodes          "Nodes"            [ stateTopic="devices/senseo/$nodes"]
Type string : implementation "Implementation"   [ stateTopic="devices/senseo/$implementation"]
Type string : interval       "Update interval"  [ stateTopic="devices/senseo/$stats/interval"]`

ThomDietrich, 2019-07-15:
Hey guys! This is great! I will try to update my own setup soon and I'll benefit from your addition there 👌  Thanks!!

Your configuration looks nice @mboremski please feel free to create a PR with as many changes as you want!! Adding the config example to the README would be great.

Did you guys see any other room how the firmware can be improved? I guess the next big step will be upgrading to homie v3. Did you by any chance already try that?

Btw what did you mean with

> Conversion of string-states "true"&"false" is a bit rude i think.

mboremski, 2019-07-16:
>Btw what did you mean with
>>Conversion of string-states "true"&"false" is a bit rude i think.

I only managed to convert the values on **items**-level in a way that it might be usefull as a **switch** in PaperUI.
I was searching for a way to convert directly on **channel**-level, but did not want to use things like rules and so on, to get real switch items (and not text-items with values useable as switch)

> free to create a PR

I'll go on with that.. :-)

ThomDietrich, 2019-07-19:
I'm not sure if I understood but the whole true/On/ON/1 discussion was also part of the Homie v3 discussions. I don't have the overview right now but this might change things.
It would be cool to update the whole codebase. Can't give any promises when I'll get to it though. Feel free to do some experiments :)

mboremski, 2019-07-20:
For me this true/false <-> ON/OFF discussion also is a bit strange.
I dont understand why openHAB cant take true/false values into a switch-item.
(if openHAB is able to I did not find the solution)

If you also dont know know how to solve in openHAB maybe it might be easier to solve in the codebase here.
I'll post an issue to adress this, so we could close this one if the config-example is merged to the docs. ;-)

ThomDietrich, 2019-07-22:
Merged example. Thanks

closed 2019-07-22 by ThomDietrich