ThomDietrich/SenseoWifi #20: Needs: platform = espressif8266@2.0.4

mboremski, 2019-07-16:
Got the following error:
```
✔ MQTT ready
Triggering MQTT_READY event...
Calling setup function...
ISR not in IRAM!
Abort called

>>>stack>>>
ctx: cont
sp: 3ffffce0 end: 3fffffc0 offset: 01b0
3ffffe90:  000000f4 00000001 3ffe86a9 00000001
3ffffea0:  3ffefa60 000001f4 0000000f 3fff0f60
3ffffeb0:  3ffefa60 3ffef5e0 3ffef3c8 40100932
3ffffec0:  3ffefa60 3ffef8b4 3ffef3c8 402015b2
3ffffed0:  3ffefa60 3ffef5e0 3ffef5e0 4020d6b0
3ffffee0:  4026099d 0000002e 3fff0f9c 40227d44
3ffffef0:  00000000 3ffef5e0 00000018 3fff0f60
3fffff00:  3ffefa60 3ffef5e0 3ffef3c8 402278aa
3fffff10:  3fff00cc 000003bd 3ffef3c8 401000c5
3fffff20:  3ffefa60 3ffef5e0 3ffef3c8 402085f9
3fffff30:  3fff00a8 00000000 0000000f 402315f6
3fffff40:  3fff2470 00000000 3fff04e8 40220dfd
3fffff50:  3ffef2ec 00000000 3fff246c 40227904
3fffff60:  3ffef32c 00000000 3fff246c 402129df
3fffff70:  3fff2484 40231636 40220dc0 3ffefe3c
3fffff80:  3fffdad0 00000000 3ffef5b8 4020283a
3fffff90:  00000000 00000000 3ffefe0c 40202350
3fffffa0:  3fffdad0 00000000 3ffefe0c 4020fdf8
3fffffb0:  feefeffe feefeffe 3ffe859c 40100c05
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v8b899c12
~ld
```

as described [here](https://github.com/esp8266/Arduino/issues/6127) this needs to be fixed with:
`platform = espressif8266@2.0.4`
in platformio.ini (for now)

ThomDietrich, 2019-07-19:
Do you already know how we need to solve this in the long run?

mboremski, 2019-07-20:
Unfortunately not.
I dont really understand the details whats wrong here.
Maybe we neet to wait until it is solved in espressif8266 ?
I am not habile enough to find a solution for us here.

ThomDietrich, 2020-02-08:
Fixed with the upgrade to Homie v3

closed 2020-02-08 by ThomDietrich