ThomDietrich/SenseoWifi #22: Homebridge?

solbadguy2010, 2019-07-17:
Hi,
i really would like to get my Senso in Homekit by unsing Homebridge.
Is this possible? Could somebody help me with this?

Existing Homebridge is running and Hardware modifications are not a problem. Software is the part i am bad at :-)

ThomDietrich, 2019-07-19:
I think you can connect Homebridge with openHAB and an openHAB connection with the machine is documented. Not sure if this would be the best or easiest or feasible solution for you

solbadguy2010, 2019-07-20:
Hi,
I think this would work in some way. Every openHAB button would be an Single Homekit object. The idea behind an native homebridge plugin is, that you than have one homekit Object with all the functionality.

closed 2020-02-08 by ThomDietrich