ThomDietrich/SenseoWifi #24: Stop brewing if cup is removed

mboremski, 2019-07-20:
I tried out what happens if the cup is removed while coffee is brewed.
(Maybe cup falls of ore someone is in hurry.. ;-) )

Maybe someone can implement a feature that switches the machine off in such cases?

ThomDietrich, 2019-07-22:
Interesting edge case, I'm not sure if we want to actively react to this event. If the user removes the cup they may just as well turn the machine off.

ThomDietrich, 2020-02-08:
@mboremski want to provide a PR? P.S. I've updated the code to Homie 3, check it out.

ThomDietrich, 2021-10-14:
Hey, just in case you are still using the machine. I did push some considerable updates the last couple of days.
Regarding the request here, I feel this is not as useful. If the user decides to remove the cup, they probably know what they are doing...

closed 2021-10-14 by ThomDietrich