ThomDietrich/SenseoWifi #26: Problem beim compelieren

diskus-tempel, 2020-02-04:
Hello I try the d1zu flashing but without success he has problems with the firmware

_-----------------------
GED_FW_NAME, __FLAGGED_FW_VERSION);

                                                                                             ^
src \ SenseoWifi.cpp: 352: 3: note: in expansion of macro 'Homie_setFirmware'
   Homie_setFirmware ("senseo-wifi-wemos", senseoWifiFWVersion);
   ^
*** [.pio \ build \ d1_mini \ src \ SenseoWifi.cpp.o] Error 1

I'm not getting anywhere, there is a tip for me here

ThomDietrich, 2020-02-04:
Hallo Diskus-tempel,
wenn du möchtest können wir gern auf Deutsch reden. 
Dein Ausschnitt reicht mir leider noch nicht. Kannst bitte die komplette Ausgabe hier her kopieren?

Normalerweise stammt ein solcher Fehler immer von einer fehlerhaften Konfiguration ab oder es werden die falschen Versionen der gelinkten Bibliotheken verwendet. Kannst du bitte deine Bibliotheken prüfen? Poste gern so viel wie du kannst hier.

Ergänzung: Vermutlich liegt es an der Homie Version. Ich muss den Code hier dringend auf den aktuellen Stand bringen.
Probiere bitte das folgende in platformio.ini:
```ini
lib_deps = https://github.com/homieiot/homie-esp8266.git#v2.2
```

diskus-tempel, 2020-02-04:
Hi , gut das ich auch in deutsch schreiben kann , 
ich habe in die ini nun folgendes rein 
`lib_deps = https://github.com/homieiot/homie-esp8266.git`
nun Fehlert das compelieren an anderen stellen 

`> Executing task: C:\Users\pcwasser\.platformio\penv\Scripts\platformio.exe run <

Processing d1_mini (platform: espressif8266@2.0.4; board: d1_mini; framework: arduino)
-------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/d1_mini.html
PLATFORM: Espressif 8266 2.0.4 > WeMos D1 R2 and mini
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES: toolchain-xtensa 1.40802.0 (4.8.2), framework-arduinoespressif8266 2.20500.190223 (2.5.0), tool-esptool 1.413.0 (4.13)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Unicode decode error has occurred, please remove invalid (non-ASCII or non-UTF8) characters from C:\Users\pcwasser\.platformio\packages\framework-arduinoespressif8266\libraries\ESP8266HTTPUpdateServer\library.properties file
Unicode decode error has occurred, please remove invalid (non-ASCII or non-UTF8) characters from C:\Users\pcwasser\.platformio\packages\framework-arduinoespressif8266\libraries\ESP8266HTTPUpdateServer\library.properties file
LibraryManager: Installing homie-esp8266
git version 2.25.0.windows.1
Cloning into 'c:\Users\pcwasser\Desktop\SenseoWifi-master\SenseoWifi-master\.pio\libdeps\d1_mini\_tmp_installing-0chtt7q7-package'...
PermissionError: [WinError 5] Zugriff verweigert: 'c:\\Users\\pcwasser\\Desktop\\SenseoWifi-master\\SenseoWifi-master\\.pio\\libdeps\\d1_mini\\_tmp_installing-0chtt7q7-package\\.git\\objects\\pack\\pack-572729e78668a4b9d3c938efeab33fe4b556aed4.idx':
  File "C:\Users\pcwasser\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 151:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\pcwasser\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 605:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\pcwasser\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 286:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\pcwasser\.platformio\platforms\espressif8266\builder\main.py", line 267:
    target_elf = env.BuildProgram()
  File "C:\Users\pcwasser\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
    return self.method(*nargs, **kwargs)
  File "C:\Users\pcwasser\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 140:
    _build_project_deps(env)
  File "C:\Users\pcwasser\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 48:
    project_lib_builder = env.ConfigureProjectLibBuilder()
  File "C:\Users\pcwasser\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
    return self.method(*nargs, **kwargs)
  File "C:\Users\pcwasser\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 1027:
    project.install_dependencies()
  File "C:\Users\pcwasser\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 859:
    lm.install(uri)
  File "C:\Users\pcwasser\.platformio\penv\lib\site-packages\platformio\managers\lib.py", line 301:
    force=force,
  File "C:\Users\pcwasser\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 696:
    pkg_dir = self._install_from_url(name, url, requirements, track=True)
  File "C:\Users\pcwasser\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 497:
    return self._install_from_tmp_dir(_tmp_dir, requirements)
  File "C:\Users\pcwasser\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 583:
    shutil.move(tmp_dir, pkg_dir)
  File "C:\Users\pcwasser\.platformio\python37\lib\shutil.py", line 578:
    rmtree(src)
  File "C:\Users\pcwasser\.platformio\python37\lib\shutil.py", line 516:
    return _rmtree_unsafe(path, onerror)
  File "C:\Users\pcwasser\.platformio\python37\lib\shutil.py", line 395:
    _rmtree_unsafe(fullname, onerror)
  File "C:\Users\pcwasser\.platformio\python37\lib\shutil.py", line 395:
    _rmtree_unsafe(fullname, onerror)
  File "C:\Users\pcwasser\.platformio\python37\lib\shutil.py", line 395:
    _rmtree_unsafe(fullname, onerror)
  File "C:\Users\pcwasser\.platformio\python37\lib\shutil.py", line 400:
    onerror(os.unlink, fullname, sys.exc_info())
  File "C:\Users\pcwasser\.platformio\python37\lib\shutil.py", line 398:
    os.unlink(fullname)
============================================== [FAILED] Took 5.42 seconds ==============================================
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.`

in der fehlerhaften datei ist folgende zeile
`HomieNode senseoNode("machine", "machine");`

[image]

diskus-tempel, 2020-02-04:
und mit dem alten github link macht er folgendes 

`> Executing task: C:\Users\pcwasser\.platformio\penv\Scripts\platformio.exe run --target upload <

Processing d1_mini (platform: espressif8266@2.0.4; board: d1_mini; framework: arduino)
-------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/d1_mini.html
PLATFORM: Espressif 8266 2.0.4 > WeMos D1 R2 and mini
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES: toolchain-xtensa 1.40802.0 (4.8.2), framework-arduinoespressif8266 2.20500.190223 (2.5.0), tool-esptool 1.413.0 (4.13), tool-espotapy 1.1.3, tool-mkspiffs 1.200.0 (2.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Unicode decode error has occurred, please remove invalid (non-ASCII or non-UTF8) characters from C:\Users\pcwasser\.platformio\packages\framework-arduinoespressif8266\libraries\ESP8266HTTPUpdateServer\library.properties file
Found 34 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <Homie> 3.0.0 #cfb3346
|   |-- <ArduinoJson> 6.14.1
|   |-- <AsyncMqttClient> 0.8.1
|   |   |-- <ESPAsyncTCP> 1.2.2
|   |   |   |-- <ESP8266WiFi> 1.0
|   |-- <Bounce2> 2.52
|   |-- <ESP Async WebServer> 1.2.3
|   |   |-- <ESPAsyncTCP> 1.2.2
|   |   |   |-- <ESP8266WiFi> 1.0
|   |   |-- <Hash> 1.0
|   |   |-- <ESP8266WiFi> 1.0
|   |   |-- <ArduinoJson> 6.14.1
|   |-- <Ticker> 1.0
|   |-- <ESP8266WiFi> 1.0
|   |-- <DNSServer> 1.1.1
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <ESP8266HTTPClient> 1.2
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <ESPAsyncTCP> 1.2.2
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <ESP8266mDNS> 1.2
|   |   |-- <ESP8266WiFi> 1.0
Building in release mode
Compiling .pio\build\d1_mini\src\Cup.cpp.o
Compiling .pio\build\d1_mini\src\SenseoControl.cpp.o
Compiling .pio\build\d1_mini\src\SenseoLed.cpp.o
Compiling .pio\build\d1_mini\src\SenseoSM.cpp.o
Compiling .pio\build\d1_mini\src\SenseoWifi.cpp.o
Compiling .pio\build\d1_mini\src\testIO.cpp.o
Generating LD script .pio\build\d1_mini\ld\local.eagle.app.v6.common.ld
Archiving .pio\build\d1_mini\libbe3\libArduinoJson-6.14.1.a
Compiling .pio\build\d1_mini\libbd4\ESP8266WiFi\BearSSLHelpers.cpp.o
Compiling .pio\build\d1_mini\libbd4\ESP8266WiFi\CertStoreBearSSL.cpp.o
Compiling .pio\build\d1_mini\libbd4\ESP8266WiFi\ESP8266WiFi.cpp.o
src\SenseoWifi.cpp:22:42: error: no matching function for call to 'HomieNode::HomieNode(const char [8], const char [8])'
 HomieNode senseoNode("machine", "machine");
                                          ^
src\SenseoWifi.cpp:22:42: note: candidates are:
In file included from .pio\libdeps\d1_mini\Homie@src-f547bfee2e673939917c59984b331ba9\src/Homie/Datatypes/Interface.hpp:11:0,
                 from .pio\libdeps\d1_mini\Homie@src-f547bfee2e673939917c59984b331ba9\src/Homie.hpp:6,
                 from .pio\libdeps\d1_mini\Homie@src-f547bfee2e673939917c59984b331ba9\src/Homie.h:4,
                 from src\SenseoWifi.cpp:7:
.pio\libdeps\d1_mini\Homie@src-f547bfee2e673939917c59984b331ba9\src/Homie/Datatypes/../../HomieNode.hpp:81:3: note: HomieNode::HomieNode(const char*, const char*, const char*, bool, uint16_t, uint16_t, const NodeInputHandler&)
   HomieNode(const char* id, const char* name, const char* type, bool range = false, uint16_t lower = 0, uint16_t upper = 0, const HomieInternals::NodeInputHandler& nodeInputHandler = [](const HomieRange& range, const String& property, const
String& value) { return false; });
   ^
.pio\libdeps\d1_mini\Homie@src-f547bfee2e673939917c59984b331ba9\src/Homie/Datatypes/../../HomieNode.hpp:81:3: note:   candidate expects 7 arguments, 2 provided
.pio\libdeps\d1_mini\Homie@src-f547bfee2e673939917c59984b331ba9\src/Homie/Datatypes/../../HomieNode.hpp:75:7: note: HomieNode::HomieNode(const HomieNode&)
 class HomieNode {
       ^
.pio\libdeps\d1_mini\Homie@src-f547bfee2e673939917c59984b331ba9\src/Homie/Datatypes/../../HomieNode.hpp:75:7: note:   candidate expects 1 argument, 2 provided
src\SenseoWifi.cpp: In function 'void senseoStateExitAction()':
src\SenseoWifi.cpp:126:10: warning: enumeration value 'SENSEO_unknown' not handled in switch [-Wswitch]
   switch (mySenseoSM.getStatePrev()) {
          ^
src\SenseoWifi.cpp: In function 'void senseoStateEntryAction()':
src\SenseoWifi.cpp:198:10: warning: enumeration value 'SENSEO_unknown' not handled in switch [-Wswitch]
   switch (mySenseoSM.getState()) {
          ^
In file included from .pio\libdeps\d1_mini\Homie@src-f547bfee2e673939917c59984b331ba9\src/Homie.h:4:0,
                 from src\SenseoWifi.cpp:7:
src\SenseoWifi.cpp: In function 'void setup()':
.pio\libdeps\d1_mini\Homie@src-f547bfee2e673939917c59984b331ba9\src/Homie.hpp:27:214: warning: statement has no effect [-Wunused-value]
 #define Homie_setFirmware(name, version) const char* __FLAGGED_FW_NAME = "\xbf\x84\xe4\x13\x54" name "\x93\x44\x6b\xa7\x75"; const char* __FLAGGED_FW_VERSION = "\x6a\x3f\x3e\x0e\xe1" version "\xb0\x30\x48\xd4\x1a"; Homie.__setFirmware(__FLAGGED_FW_NAME, __FLAGGED_FW_VERSION);

                                                                                             ^
src\SenseoWifi.cpp:352:3: note: in expansion of macro 'Homie_setFirmware'
   Homie_setFirmware("senseo-wifi-wemos", senseoWifiFWVersion);
   ^
Compiling .pio\build\d1_mini\libbd4\ESP8266WiFi\ESP8266WiFiAP.cpp.o
Compiling .pio\build\d1_mini\libbd4\ESP8266WiFi\ESP8266WiFiGeneric.cpp.o
*** [.pio\build\d1_mini\src\SenseoWifi.cpp.o] Error 1
============================================== [FAILED] Took 6.95 seconds ==============================================
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.
`

Liebe grüße Sven

diskus-tempel, 2020-02-04:
Jetzt bin ich weiter gekommen .. 
dennoch probleme , 
das erste lag an der arduinojson version 

nun bekomm ich es auf geflasht jedoch verbindet es sich nicht mit mein wlan .. 
wenn ich auf den esp will sagt er ui bundle not loadet . see configuration api usage:

in dem serialen monitor hab ich folgendes 
`> Executing task: C:\Users\pcwasser\.platformio\penv\Scripts\platformio.exe device monitor <

--- Miniterm on COM3  115200,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
{l␀l��|␀�d�|␃␌␄␌�␄l�␌c|��␃�␓�;�c�␄c��g'�lno���␌b␜x��l{l{$p�'�␐␂␄␌�␄l␄��␄␌␄c␌n�|␃$�␄␄�b��og�␀l��l ␃�␛␓no␄$`␃␏␃gs���o␌␄#␄�␎d␏;��g␄␄c␄�␏l�␒d�p␃���␛�`␃��o�␃✖ /homie/config.json doesn't exist
Configuration invalid. Using CONFIG MODE
💡 Firmware senseo-wifi-wemos ()
🔌 Booting into config mode 🔌
Device ID is ecfabc58a2e4
AP started as SenseoWifi-ecfabc58a2e4 with IP 192.168.123.1
Triggering Wi-Fi scan...`

rmickael62, 2020-02-04:
hallo, du musst zeile 22 so ändern: HomieNode senseoNode("machine", "machine"); => HomieNode senseoNode("machine", "Machine", "machine");

diskus-tempel, 2020-02-04:
ich komm gerate nicht weiter mit der configuration .. da ich auch kein webinterface erhalte 

wie oben im vorherigen beitrag schon gezeigt

ThomDietrich, 2020-02-04:
Hey, das ändern dieser einen Zeile mag deinen unmittelbaren Fehler beheben aber das wird nicht reichen. Du musst solang ich das gesamte Programm nicht angepasst die alte Homie version nutzen. Aus deinem Consolentext geht auch hervor dass du noch die neue Homie Version statt version 2.2:

> Dependency Graph
> |-- <Homie> 3.0.0 #cfb3346

Wenn du das gelöst bekommst sollte es klappen. Belies dich dazu mal auf der PlatformIO Seite. Wenn nicht kannst du darauf warten dass ich den Code auf Homie 3 umstelle. Ich kann aber nicht sagen wie schnell ich as hinbekomme. Könnte bereits am Samstag klappen oder drei Wochen dauern 🙃 

Nebenbei: Wie weit bist du mit dem Umbau? Ging alles problemlos?

diskus-tempel, 2020-02-04:
Ich werde versuchen das hinzubekommen,  
Jedoch kann ich auch warten bis du es schafst..  Ich hab Ja wenig bis keine erfahrung mit dem. Programmierung UMD auch mit den genutzen softwaren 

Die platine denke ich ist weniger das problem.  

Das ist eher meine welt. 
Hab bis auf ground alles verlötet auf einer lochraster platine.  Mit optukopller 
Wenn der d1 läuft wird er ausgelötet.  Und es geht an den praxis test :)

Ich freu mich schon wenn es denn mal läuft.  
Würde mir den start in den tag etwas erleichtern :)

diskus-tempel, 2020-02-04:
Würde evtl wenn der test klappt au gerne von euch so n bord kaufen da es dich sehr sauber aussieht

diskus-tempel, 2020-02-04:
[image]
[image]

diskus-tempel, 2020-02-04:
Das sind derzeit meine probleme

ThomDietrich, 2020-02-05:
Gut dann blocke ich mir wohl für Samstag mal etwas Zeit. Machen wollte ich die Umstellung schon lange, aber die Zeit muss ich mir natürlich nehmen.

Deine Screenshots sehen sehr gut aus! Du müsstest jetzt den Homie Schritten folgen oder besser du flashst das optionale UI bundle. Dazu kann ich am Wochenende etwas mehr dokumentieren. Siehe hier: https://github.com/homieiot/homie-esp8266/tree/develop/data/homie

Die Platine können wir dir gern zuschicken! Ich hab die damals im Dutzend günstiger bekommen ;) Schick doch bitte deine Postanschrift an blindfuchs@nurzen.de und wir bringen das auf den Weg. Fehlt dir noch eine Komponente die wir mit beilegen sollen?
Statt über "kaufen" zu reden würde ich übrigens vorschlagen dass du uns im Gegenzug dein ausfühliches Feedback mit Kritik und Verbesserungsvorschlägen gibst 😉

diskus-tempel, 2020-02-05:
Hi,  wenn ihr noch einen Wemos habt und so optukopller,  diese musste ich so aus dem netz bezihen,  

Wenn kein verkauf dann aba ein danke fur die kaffeekasse dafur wäre eure paypal adresse ganz gut :) wäre super wenn du es hinbekommst.  

Dann kann ich auch nochmal neu anfangen,  evtl eine bild anleitung mit dazu machen wenn es euch hilft und wie ich zufrieden bin bez feedback würde ich eh schreiben,  Zumal au direkt hilfe kommt!  

Adresse wäre 
Sven brendel 
Kaufbeurer str 1
90455 nürnberg 

Werde evtl heute abend versuchen das ich mal weiter komme!  

Lg und einen Angenehmen tag 
--
Diese Nachricht wurde von meinem Android Mobiltelefon mit GMX Mail gesendet.

diskus-tempel, 2020-02-05:
guten abend zusammen 
ich habe nun wieder einige stunden damit verbracht ich komme hier nicht weiter derzeit :( 

ich habe das UI-Bundle auch in die verzeichnisse rein geladen . 
ob das spiff uploade ist oder nicht weis ich leider nicht ! 

ich hab meinen wemos nun bestimmt zum 500 mal geflasht . und ich habe immer das selbe ergebnis

ich habe homie version 2.0 gefunden auch diese ohne erfolg :( 

lg sven

diskus-tempel, 2020-02-05:
email ist raus

ThomDietrich, 2020-02-08:
Hey Sven,
ich habe das Projekt einmal mehr oder weniger komplett überarbeitet. Die Firmware setzt jetzt auf Homie 3 und alles ist entsprechend angepasst. Ich hab auch eine gute Stunde in die README investiert um das Flashen mit PlatformIO zu beschreiben. Bin gespannt ob du damit klar kommst.

Die Platine ging übrigens am Donnerstag raus. Hast du sie bereits erhalten?

diskus-tempel, 2020-02-08:
Hi 

Cool werde ich wenn ich es schaffe morgenabend mal ansehen,  bin heute nicht mehr zuhause.  

Die platine ist heute gekommen :) 

Evtl schaff ich es nachher mal die readme zu lassen.  
Sag dir aber nochmal bescheid,  also wenn ich es morgen schaf!  

Lg und danke das du fur mich alles umgestrikt hast.

diskus-tempel, 2020-02-09:
Hi,  hab jetzt wieder versucht die software /firmware zu laufen auf meinen d1 hab wieder das problem das es unterlagen zum ap nicht geht :(
--
Diese Nachricht wurde von meinem Android Mobiltelefon mit GMX Mail gesendet.

diskus-tempel, 2020-02-09:
Guten abend herr dietrich,. 

Ich habe es jetzt auf 4 computer versucht. Und auch mit der neuen version,  
Ich bekomm dennoch den ui bundle fehler :( 

Ich finde den fehler nicht wo ran es liegt. 

Gibt es eine datei um den config modus zu stoppen und eine wlan datei um das zum um gehen.  Oder haben sie die moglichkeit mir via teamwiever zu helfen?  

Ich bin da langsam an den punkt wo ich nicht mehr weiter weis,  da ich sogar auf mehreren computern immer den selben fehler bekomm :( 

Lg sven 
Guten start in die neue woche!  
--
Diese Nachricht wurde von meinem Android Mobiltelefon mit GMX Mail gesendet.

ThomDietrich, 2020-02-10:
Hallo,
ruf mich doch bitte gegen 19 Uhr an. 0173/6152492

ThomDietrich, 2021-10-04:
Hallo,
ich habe heute ein paar Kleinigkeiten verändert und das Kompilieren und Hochladen erfolgreich ohne Probleme hinbekommen. Wollt ich nur erwähnen falls es dich interessiert ;) lg

closed 2021-10-04 by ThomDietrich