ThomDietrich/SenseoWifi #29: Arduino Core fix

ThomDietrich, 2020-05-07:
@pattyland  wrote in https://github.com/ThomDietrich/SenseoWifi/pull/28:

> Hi,
> 
> there seems to be [problems with newer version of arduino core](https://github.com/esp8266/Arduino/issues/6127). 
> My D1 Mini would crash at boot with a `ISR not in IRAM` error. 
> 
> This PR ist the [quick fix](https://github.com/esp8266/Arduino/issues/6127), [this is the longer (and better?) fix](https://github.com/esp8266/Arduino/issues/6127#issuecomment-497785103) I guess.
> 
> Best regards
> Sören

ThomDietrich, 2020-05-07:
@pattyland did you try the `ICACHE_RAM_ATTR` workaround?

pattyland, 2020-05-08:
No, it was late at night and pinning the platform version was quicker 🙈

PR #30 (2020-05-11): Changed Led interrupt handling to ICACHE_RAM_ATTR. (#29)
Changed Led interrupt handling to ICACHE_RAM_ATTR. Works with actual espressi8266 2.7.1. 
Changed platformio.ini to get actual version of espressi8266 

Fixes #29
closed 2020-05-11 by ThomDietrich