ThomDietrich/SenseoWifi #35: einbindung im iobrocker

diskus-tempel, 2020-07-30:
Hallo zusammen , 
ich möchte mich bei Thomas Bedanken für die Platine und die hilfe bei der Kompelierung 

da ich den iobrocker als smarthomezentrale nutze habe ich mir ein blockly script gemacht um die kaffemaschiene via alexa auch anzusteuern 

hier erst einmal das script selbst 
`<xml xmlns="https://developers.google.com/blockly/xml">
  <block type="on" id=":tKFC7{i^_z~*f@]FM/6" x="88" y="67">
    <field name="OID">0_userdata.0.Kaffe</field>
    <field name="CONDITION">ne</field>
    <field name="ACK_CONDITION"></field>
    <statement name="STATEMENT">
      <block type="controls_if" id="@4da8+G1gocm1o-xyJQ:">
        <value name="IF0">
          <block type="logic_compare" id="%:#wI=a%(V4At8qa86N7">
            <field name="OP">EQ</field>
            <value name="A">
              <block type="get_value" id="S?%%m%*qZeT+OZrlRw7C">
                <field name="ATTR">val</field>
                <field name="OID">0_userdata.0.Kaffe</field>
              </block>
            </value>
            <value name="B">
              <block type="logic_boolean" id="!wB2kNZ0jepw?BN{Yax^">
                <field name="BOOL">TRUE</field>
              </block>
            </value>
          </block>
        </value>
        <statement name="DO0">
          <block type="control" id="K_KnE$};v:IUH(KO-j/t">
            <mutation xmlns="http://www.w3.org/1999/xhtml" delay_input="false"></mutation>
            <field name="OID">mqtt.0.homie.ecfabc58a2e4.machine.power.set</field>
            <field name="WITH_DELAY">FALSE</field>
            <value name="VALUE">
              <block type="text" id="..i4;OBa9)%m4XX5]=#a">
                <field name="TEXT">true</field>
              </block>
            </value>
            <next>
              <block type="controls_if" id="%=o27YD4dG=8d~aRrH2S" disabled="true">
                <value name="IF0">
                  <block type="logic_compare" id="YBjf-q(O@A;dg:~%mMVv">
                    <field name="OP">EQ</field>
                    <value name="A">
                      <block type="get_value" id="rBh%r{mN-APBpF~M?#8]">
                        <field name="ATTR">val</field>
                        <field name="OID">mqtt.0.homie.ecfabc58a2e4.machine.opState</field>
                      </block>
                    </value>
                    <value name="B">
                      <block type="text" id="c|:ZU(p3k35wloog,e#2">
                        <field name="TEXT">SENSEO_READY</field>
                      </block>
                    </value>
                  </block>
                </value>
                <statement name="DO0">
                  <block type="control" id="LPFW*QKvPaMp4L[nUuwl">
                    <mutation xmlns="http://www.w3.org/1999/xhtml" delay_input="false"></mutation>
                    <field name="OID">mqtt.0.homie.ecfabc58a2e4.machine.brew.set</field>
                    <field name="WITH_DELAY">FALSE</field>
                    <value name="VALUE">
                      <block type="text" id="sTv^lcL{];_~^/l_;Rl:">
                        <field name="TEXT">1cup</field>
                      </block>
                    </value>
                    <next>
                      <block type="control" id="E%YI*GT@Tx_I5TRP./%y">
                        <mutation xmlns="http://www.w3.org/1999/xhtml" delay_input="true"></mutation>
[... 157 lines cut ...]
    </next>
  </block>
</xml>
`

des weiteren müsst ihr bei objekten ein virtuellen switch anlegen und diesen mittels iot in alexa intigrieren 

und im link auf eure id´s alles abändern . 

dann geräte suchen . dann findet alexa den virtuellen switch 

und hier könnt ihr dann in alexa eine routine anlegen ! 

kleine infomation noch dazu 

wenn die senseo manuell an den tasten bedind wird . 

sollte wasser und ped bereits innen liegen da das script direkt nach fertigung des heizens auch startet zu brühen 
desweiteren habe ich im script das wenn der wassertank leer ist alexa in der küche mir das auch sagt 

folgen wird noch der ped zähler und die infomationsübertragung an telegram 

für weitere fragen einfach schreiben 

lg aus nürnberg

ymmas469, 2020-09-18:
Hallo, ich habe auch vor das Projekt in IO Broker einzubinden.
Ich empfange auch alle Daten der Maschine über mqtt.
Schaffe es allerdings nicht sie anzusprechen.
Über welches Topic gebe ich denn wohl die Befehle?
Würde mich sehr über etwas Hilfe freuen.
lg

diskus-tempel, 2020-09-22:
Hi gerade erst gesehen , kann dir das später bildlich Mal zeigen , weibei topic usw dem Script auch zu entnehmen sind ! LG --Diese Nachricht wurde von meinem Android Mobiltelefon mit GMX Mail gesendet.

ymmas469, 2020-09-22:
Hey, das wäre spitze, habe auch versucht dein Skript einzufügen, das mache ich aber vermutlich auch schon falsch. Ich bin noch ein blutiger Anfänger was das ganze Thema betrifft.

diskus-tempel, 2020-09-22:
wenn du mir mal deine email adresse gibst kann ich dir mal meine tel nr senden telefonisch is das alles etwas leichter zu erklären , diesen supoort hatte ich auch bekommen und möchte ich auch weiter geben

ymmas469, 2020-09-22:
Wow, das ist nett.
bernhardbark9.bb@gmail.com

diskus-tempel, 2020-09-22:
mail is raus an dich !

ThomDietrich, 2021-11-14:
Hey guys, closing this thread.

Just wanted to make you aware that I have recently pushed a ton of changes to this project. There are a few really good improvements and new features. Check it out if you like.

closed 2021-11-14 by ThomDietrich