ThomDietrich/SenseoWifi #36: Build endet mit Abhängigkeitsfehlern

MichiW1, 2020-08-11:
Hallo ThomDietrich,

du hast ein tolles Projekt hier, das ich unbedingt nachbauen wollte.
Leider habe ich nicht ganz so tiefe Programmierkenntnisse, habe aber deine Anleitung Schritt für Schritt durchgearbeitet.

Wenn ich im VS Code den Build über PlatformIO starte, kommt folgender Fehler.
Ich habe keine Ahnung, warum die Abhängigkeiten der Libraries nicht richtig aufgelöst werden... habe alles versucht und bin am Verzweifeln. Was mache ich falsch?

> Executing task in folder SenseoWifi: C:\Users\HP\.platformio\penv\Scripts\platformio.exe run <

Processing senseo-wifi (platform: espressif8266; board: d1_mini; framework: arduino)
-------------------------------------------------------------------------------------------------------------------------------

Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/d1_mini.html
PLATFORM: Espressif 8266 2.6.2 > WeMos D1 R2 and mini
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES:
 - framework-arduinoespressif8266 3.20704.0 (2.7.4)  
 - tool-esptool 1.413.0 (4.13)
 - tool-esptoolpy 1.20800.0 (2.8.0)
 - toolchain-xtensa 2.40802.200502 (4.8.2)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 36 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <Homie> 3.0.0 #cfb3346  
|   |-- <ArduinoJson> 6.16.1
|   |-- <AsyncMqttClient> 0.8.2
|   |   |-- <ESPAsyncTCP> 1.2.2
|   |   |   |-- <ESP8266WiFi> 1.0
|   |-- <Bounce2> 2.55
|   |-- <ESP Async WebServer> 1.2.3
|   |   |-- <ESPAsyncTCP> 1.2.2
|   |   |   |-- <ESP8266WiFi> 1.0
|   |   |-- <Hash> 1.0
|   |   |-- <ESP8266WiFi> 1.0
|   |   |-- <ArduinoJson> 6.16.1
|   |-- <Ticker> 1.0
|   |-- <ESP8266WiFi> 1.0
|   |-- <DNSServer> 1.1.1
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <ESP8266HTTPClient> 1.2
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <ESPAsyncTCP> 1.2.2
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <ESP8266mDNS> 1.2
|   |   |-- <ESP8266WiFi> 1.0
Building in release mode
Compiling .pio\build\senseo-wifi\src\Cup.cpp.o
Compiling .pio\build\senseo-wifi\src\SenseoControl.cpp.o
Compiling .pio\build\senseo-wifi\src\SenseoLed.cpp.o
Compiling .pio\build\senseo-wifi\src\SenseoSM.cpp.o
In file included from C:\Users\HP\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/Arduino.h:238:0,
                 from .pio\libdeps\senseo-wifi\Homie\src/Homie.hpp:3,
                 from .pio\libdeps\senseo-wifi\Homie\src/Homie.h:4,
                 from src\Cup.h:9,
                 from src\Cup.cpp:7:
c:\users\hp\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\4.8.2\algorithm:60:30: fatal error: utility: No 
such file or directory
 #include <utility> // UK-300.
                              ^
compilation terminated.
In file included from C:\Users\HP\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/Arduino.h:238:0,
                 from .pio\libdeps\senseo-wifi\Homie\src/Homie.hpp:3,
                 from .pio\libdeps\senseo-wifi\Homie\src/Homie.h:4,
                 from src\SenseoControl.h:9,
                 from src\SenseoControl.cpp:7:
c:\users\hp\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\4.8.2\algorithm:60:30: fatal error: utility: No 
such file or directory
 #include <utility> // UK-300.
                              ^
compilation terminated.
*** [.pio\build\senseo-wifi\src\Cup.cpp.o] Error 1
In file included from C:\Users\HP\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/Arduino.h:238:0,
                 from .pio\libdeps\senseo-wifi\Homie\src/Homie.hpp:3,
                 from .pio\libdeps\senseo-wifi\Homie\src/Homie.h:4,
                 from src\SenseoLed.h:9,
                 from src\SenseoLed.cpp:7:
c:\users\hp\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\4.8.2\algorithm:60:30: fatal error: utility: No 
such file or directory
 #include <utility> // UK-300.
                              ^
compilation terminated.
In file included from C:\Users\HP\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/Arduino.h:238:0,
                 from .pio\libdeps\senseo-wifi\Homie\src/Homie.hpp:3,
                 from .pio\libdeps\senseo-wifi\Homie\src/Homie.h:4,
                 from src\SenseoSM.h:9,
                 from src\SenseoSM.cpp:7:
c:\users\hp\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\4.8.2\algorithm:60:30: fatal error: utility: No 
such file or directory
 #include <utility> // UK-300.
                              ^
compilation terminated.
*** [.pio\build\senseo-wifi\src\SenseoControl.cpp.o] Error 1
*** [.pio\build\senseo-wifi\src\SenseoLed.cpp.o] Error 1
*** [.pio\build\senseo-wifi\src\SenseoSM.cpp.o] Error 1
================================================= [FAILED] Took 7.87 seconds =================================================
The terminal process "C:\Users\HP\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1.

Danke für deine Hilfe,
Michael

MichiW1, 2020-08-24:
Hallo zusammen,
ich habe mein Problem nun selbst lösen können:

Es gibt wohl einen größeren Fehler bei der Platform.io extension für MS VSCode unter einem Windows Rechner. Einige Dateien und Verzeichnisse werden bei der Installation nicht angelegt. Unter Linux läuft es hingegen perfekt.

Konkret geht es um die Inhalte im Verzeichnis: 
C:\Users\{username}\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\4.8.2
Ich habe diesen Ordner einfach aus der erfolgreichen Linux-Installation (liegt dort im home-Verzeichnis unter gleicher Struktur) mit allen Inhalten in den Windows-Ordner kopiert und schon läuft der Build durch... 

Sicher keine schöne Lösung, aber es geht. 
(Neuinstallation von platform.io unter Windows etc. hatte ich vorher alles vergeblich versucht :-()

Grüße, Michael

closed 2020-08-24 by MichiW1

ThomDietrich, 2020-08-24:
Hey Michael! Freut mich dass du das Problem lösen konntest und auch für den nächsten Nutzer mit dem Problem dokumentiert hast. Danke dafür. Viele Grüße!