ThomDietrich/SenseoWifi #39: Compatibility question

kennhardy, 2020-12-14:
Hi, would love to test this modification on my Sensei Switch HD6594/60. Do you think it might work? Mostly the electronics / schematis are needed to work. If code needs adjusting I might hopefully figure those out.

ThomDietrich, 2020-12-16:
Hey!
I had a quick look at the manual. The machine seems to have three buttons for the three different modes. Furthermore it has a "Calc light". What I am not quite sure of: Does every button have it's own LED? Please let me know so we can check that enough I/O ports are available.

So here it goes. The code and instructions here can definitely be used! You need to add the additional buttons and LEDs to the circuit and modify the logic to account for the different modes. I don't see any real issues here. Also, you should be able to get one-two modes working by leveraging the existing logic right away. This way you can start testing, get it working, and go from there.

Let me know what you decide! Best! Thomas

When you start working on this project remember to be careful with the open circuit. At least in the classic Senseo, most if it is connected to AC.

kennhardy, 2020-12-16:
Hi there @ThomDietrich! Thank you for replying so promptly.

The machine has three buttons and three led's - one for each button. Each button do specific tasks based on what you want (3in1).

Buttons in order back to front:
 - Brew pot of coffee, with filter
 - Brew big cup of coffee, big cartridge used
 - Brew medium cup of coffee, small cartridge used. This button also has a double-click feature which brews a even smaller cup coffee, more like an espresso.

For me, automating the morning coffee - two buttons / features would suffice since I only fetch one cup on my way out the door. But if you'd like to support this machine with all features i'd gladly purchase the necessary parts and provide you with any needed information about it.

Have a nice one and merry christmas to you!

ThomDietrich, 2021-10-04:
Hey, sorry I never got back to you. What did you end up doing? With all the buttons and LEDs the first problem that came to mind is the lack of IO pins on the Wemos...

closed 2021-11-14 by ThomDietrich