ThomDietrich/SenseoWifi #40: Debug options

pattyland, 2021-01-03:
Hello,
After over a year and hundreds of successful brews, my SenseoWifi does not connect to the MQTT broker, but it does connect to the WLAN. 

What are my debugging options?
Is there, as with ESPHome, the possibility to view a log via WLAN?
Or a web gui? My portscans were not successful....

Kind regards
Sören

ThomDietrich, 2021-02-19:
Hey Sören!

Sorry for the late reply. Hopefully you solved this already. There is no web-frontend. Your best bet would be to connect via USB to the serial console. Alternatively you can always use the reset button on D4 to reset the configuration to default and show the setup web frontend.

pattyland, 2021-02-27:
For USB I would have had to disassemble the coffee machine again?
I solved it with the reset button and re-entered all the data. It would still be interesting to know what happened...

closed 2021-02-27 by pattyland