ThomDietrich/SenseoWifi #41: Implementing in Home Assistant

Post404, 2021-02-13:
Everything in this project has been new for me, soldering, ordering a custom PCB, dealing with firmware. I am now at the point that i'm trying to implement it into Home Assistant. It is connected to the Wifi and to the MQTT broker.
I just don't know how to "translate" the OpenHAB config to Home Assistant. Is there any example available that I can see?

ThomDietrich, 2021-02-15:
Hey @Post404,
I was happy to read that you were able to bring your Senseo to the point where you can read its status and send commands via MQTT. Correct? Congrats!

Please provide any issue you had or details that was missing from the description. I can certainly add more content to the README of this project.

----

I've also recently hooked my SenseoWifi to HomeAssistant. The current firmware does not support Home Assistant auto discovery via MQTT but that is on my low prio todo list. Till then you can use the following configuration as a starting point:

If you didn't already, activate the [packages concept](https://www.home-assistant.io/docs/configuration/packages/) in Home Assistant's `configuration.yaml` (this is not strictly linked to the below but is imho a must for all HA setups):

```yaml
homeassistant:
  ...
  packages: !include_dir_named packages
```

Now add the following configuration file under `packages/senseo_wifi.yaml`:
```yaml
input_boolean:
  senseowifi_brew_normal:
    name: "SenseoWifi Auslöser einfache Tasse brühen"
    initial: off
    icon: mdi:coffee

  senseowifi_brew_double:
    name: "SenseoWifi Auslöser große Tasse brühen"
    initial: off
    icon: mdi:coffee

switch senseowifi:
  - platform: mqtt
    name: SenseoWifi Power
    #
    state_topic: "homie/senseo-wifi-rf21/machine/power"
    state_on: "true"
    state_off: "false"
    #
    command_topic: "homie/senseo-wifi-rf21/machine/power/set"
    payload_on: "true"
    payload_off: "false"
    #
    availability_topic: "homie/senseo-wifi-rf21/$state"
    payload_available: "ready"
    #value_template: "{{ value }}"

sensor senseowifi:
  - platform: mqtt
    name: SenseoWifi Brewed size
    #
    state_topic: "homie/senseo-wifi-rf21/machine/brewedSize"
    #
    availability_topic: "homie/senseo-wifi-rf21/$state"
    payload_available: "ready"
    #value_template: "{{ value }}"

  - platform: mqtt
    name: SenseoWifi OpState
    #
    state_topic: "homie/senseo-wifi-rf21/machine/opState"
    #
    availability_topic: "homie/senseo-wifi-rf21/$state"
    payload_available: "ready"
    #value_template: "{{ value }}"

binary_sensor:
  - platform: mqtt
    name: SenseoWifi Out-of-water
    #
    state_topic: "homie/senseo-wifi-rf21/machine/outOfWater"
    payload_on: "true"
    payload_off: "false"
    #
    availability_topic: "homie/senseo-wifi-rf21/$state"
    payload_available: "ready"
    #value_template: "{{ value }}"

  - platform: mqtt
    name: SenseoWifi Brew
    #
    state_topic: "homie/senseo-wifi-rf21/machine/brew"
    payload_on: "true"
    payload_off: "false"
    #
    availability_topic: "homie/senseo-wifi-rf21/$state"
    payload_available: "ready"
    #value_template: "{{ value }}"

  - platform: mqtt
    name: SenseoWifi Cup available
    #
    state_topic: "homie/senseo-wifi-rf21/machine/cupAvailable"
    payload_on: "true"
    payload_off: "false"
    #
    availability_topic: "homie/senseo-wifi-rf21/$state"
    payload_available: "ready"
    #value_template: "{{ value }}"

  - platform: mqtt
    name: SenseoWifi Cup full
    #
    state_topic: "homie/senseo-wifi-rf21/machine/cupFull"
    payload_on: "true"
    payload_off: "false"
    #
    availability_topic: "homie/senseo-wifi-rf21/$state"
    payload_available: "ready"
    #value_template: "{{ value }}"

automation:
  - id: "1611257531404"
    alias: Regel SenseoWifi Auslöser normale Tasse brühen
    trigger:
      platform: state
      entity_id: input_boolean.senseowifi_brew_normal
      to: "on"
    action:
      - service: mqtt.publish
        data:
          topic: "homie/senseo-wifi-rf21/machine/brew/set"
          payload: "1cup"

  - id: "1611257531405"
    alias: Regel SenseoWifi Auslöser große Tasse brühen
    trigger:
      platform: state
      entity_id: input_boolean.senseowifi_brew_double
      to: "on"
    action:
      - service: mqtt.publish
        data:
          topic: "homie/senseo-wifi-rf21/machine/brew/set"
          payload: "2cup"

  - id: "1611257531406"
    alias: Regel SenseoWifi update beim Ausschalten
    trigger:
      platform: state
      entity_id: binary_sensor.senseowifi_brew
      to: "off"
    action:
      - service: input_boolean.turn_off
        data:
          entity_id: input_boolean.senseowifi_brew_normal
      - service: input_boolean.turn_off
        data:
          entity_id: input_boolean.senseowifi_brew_double

```

Replace "senseo-wifi-rf21" by whatever your machine is called on the MQTT broker.

Hope that helped. Please provide as much info back as you can

Post404, 2021-02-18:
This is excactly what I was looking for. But the Senseo is not responding to power switch or the 1 or 2 cups booleans.
With MQTT explorer I can see that the commands are being received and the states are changing.

I might have to take the Senseo appart again to check if everything is connected right. Or could it be something software sided?

ThomDietrich, 2021-02-19:
Take it one step at a time :)

1. Use `testIO.cpp` to ensure everything works physically, then switch back to the normal firmware mode
2. Observe correct state change etc on the MQTT broker (e.g. OpState should change when you turn the machine on and off) - pay attention to the "debug" MQTT topic for more details. Check this file https://github.com/ThomDietrich/SenseoWifi/blob/master/src/SenseoWifi.cpp for "setProperty" to see what's supposed to be published to MQTT when
4. With confidence that everything works so far, try sending commands via MQTT
5. Only after everything is working set up the connection with Home Assistant. It should work right away

Post404, 2021-02-23:
I used testIO.cpp and it detected that the cup sensor is not present which is correct. Do I need to adjust something in the code for that? Also while in test mode it turned the Senseo On and Off. So at the very least the wires for that are correct.
Then powering it configured and in normal mode makes it beeps twice, so it is connected to wifi and the MQTT broker.
In Supervisor, Mosquitto Broker I see the senseo connection in the log as well.
Then I added the MQTT integration to listen to the topics and be able to manualy publish a packet (is this the same as "send command"?)
When listening to the OpState it doesn't change at all and stays on SENSEO_unknown. It should be SENSEO_NOWATER when turned on.
Debug just says: brew: 19 seconds
Is there something i'm missing?

Post404, 2021-02-24:
I got turning on and off working now from the dashboard! I removed the senseo config and replaced it with yours. Changed the senseo name everywhere, but without translating. The opState is correctly updating and everything else as well. Just the 1 cup and 2 cups switches are not working. Debug says: "no or full cup present. Not executing." I could of course add the sensor, but is there a way to disable this?

ThomDietrich, 2021-02-24:
Great! Congrats! (Btw. be sure to add your voice to #42 😉  )

First of all: The cup sensor is super useful, so think about adding it. Of course, you don't have to. For that reason, I added a flag to disable the use of such a sensor. You must set/unset that flag during the initial setup (the process where you also set wifi password and MQTT credentials). Sorry, you need to reset your machine, there is no way to change the setting now.

Resetting:

* You can send a reset command to the power MQTT topic: https://github.com/ThomDietrich/SenseoWifi/blob/9264955691747e21e436271c5656ed9295bbca57/src/SenseoWifi.cpp#L57
* Or you can press the reset button for 5 seconds: https://github.com/ThomDietrich/SenseoWifi/blob/9264955691747e21e436271c5656ed9295bbca57/src/SenseoWifi.cpp#L378

My machines have the reset push button in one of the holes on the bottom of the machine. Quite useful for when you ever need to reconnect it to a new network.

Post404, 2021-02-25:
Might as well just add the sensor and the reset button. It's not that much work at this point. Thanks for helping me out and also making me understand it better!

Post404, 2021-02-28:
Oke, I added the sensor. When in test mode the number is 1024 and changed to ~60 when placing my hand in front of it.
But it is also not detected in both cases, like this:
Testing Cup Detector ... not detected (analog: 1024)
and later:
Testing Cup Detector ... not detected (analog: 59)

I connected the sensor to A0, do I also need to wire it to D0?

Post404, 2021-03-01:
I think I found it out. The sensor i ordered was a TCRT5000 without any board. After checking the versions with the board I noticed that they also have a D0 pin. I'm guessing that if I use that one it will all just work.
So I ordered 2 of those for good measure. Will update when they arive.

Post404, 2021-03-04:
Received the sensor, installed the sensor and now it works perfectly!

Again, thanks for all the help!

ThomDietrich, 2021-03-04:
I played around with the analog input but ended up just going with D0 for simplicity. Too bad you had to order a second time but seems like you got to a happy end. Great!!

I would leave this ticket open as I need to document Home Assistant integration in README.md. Any other details you would suggest adding?

Post404, 2021-03-04:
Now that you mention it, I remember being stuck with the question: "Where exectly do I solder de cables on the senseo board?".
After comparing this: https://github.com/ThomDietrich/SenseoWifi/blob/master/images/DSC09654.jpg picture
with this picture: https://github.com/ThomDietrich/SenseoWifi/blob/master/images/DSC09627.jpg
I eventually got it after a while, but it would have been clearer with a picture like this:
[image]

ThomDietrich, 2021-10-04:
@Post404 as the buttons and LEDs are in the other side of the PCB, I didn't even think about marking them :D you are of course right...

ThomDietrich, 2021-10-04:
Home Assistant example now in the README. You wouldn't happen to have a cooler configuration by now?

closed 2021-10-04 by ThomDietrich

Post404, 2021-10-05:
Yes I do actually, I went full circle. I went back to visualising the Senseo in Homeassistant Lovelace.

You'll need these images in your homeassistant "www" folder: https://we.tl/t-Zrkoq89kdD
It is now set up with a black senseo and a blue cup, but I could easily render different colors.

And put this code in a picture element card:
`type: picture-elements
elements:
  - type: conditional
    conditions:
      - entity: binary_sensor.senseowifi_cup_available
        state: 'on'
    elements:
      - type: image
        image: /local/Senseo_kopje.png
        style:
          pointer-event: none
          transform: translate(0,0)
          top: 0
          left: 0
          right: 0
          bottom: 0
      - type: conditional
        conditions:
          - entity: sensor.senseowifi_opstate
            state: SENSEO_BREWING
        elements:
          - type: image
            image: /local/Senseo_schuim.png
            style:
              pointer-event: none
              width: 17.5%
              transform: translate(0,0)
              top: 51%
              left: 32.2%
              right: 0
              bottom: 0
              animation: schuim 25s linear 1
      - type: conditional
        conditions:
          - entity: sensor.senseowifi_opstate
            state: SENSEO_BREWING
        elements:
          - type: image
            image: /local/Senseo_stralen.png
            style:
              pointer-event: none
              transform: translate(0,0)
              top: 0
              left: 0
              right: 0
              bottom: 0
              opacity: 0
              animation: uit 25s linear 1
      - type: conditional
        conditions:
          - entity: sensor.senseowifi_opstate
            state: SENSEO_BREWING
        elements:
          - type: image
            image: /local/Senseo_schuim_voor.png
            style:
              pointer-event: none
              width: 17.5%
              transform: translate(0,0)
              top: 51%
              left: 32.2%
              right: 0
              bottom: 0
              animation: schuim 25s linear 1
[... 96 lines cut ...]
    50% { opacity: 0; }
    93% { opacity: 0; }
    100% { opacity: 1; }
  }
  @keyframes schuim {
    0% { top: 60%; }
    0% { width: 17%; }
    0% { left: 32.3%; }
    100% { left: 32.2%; }
    100% { width: 17.5%; }
    100% { top: 51%; }
  }
  @keyframes uit {
    0% { opacity: 1; }
    98% { opacity: 1; }
    100% { opacity: 0; }
  }`

This should visualise every state even with some animation. So the cup will appear when placing it (with reflections on the senseo itself, so proud). The button light blinks correctly for every state. And when it starts brewing you see the coffee comming out and the cup starts to fill with the official Senseo coffee™. The only thing I still want to do is swap it out for a better 3D model of the Senseo.

I also added an automation in Node-red to change a light of something else depending on the state.
So: heating > orange, ready > green, brewing > blinking between orange and green and when off >returning to previous state. Blnking time is assuming 1 cup, so it blinks a little short with 2 cups.

Might need a lot of adjustment to make it work for anybody else but this is what I use:
`[{"id":"7809448eb5724189","type":"subflow","name":"Verlichting Woonkamer Check","info":"","category":"","in":[{"x":80,"y":140,"wires":[{"id":"5ab77eb286d26d9a"},{"id":"9cce0071663acf02"},{"id":"b076cebd74637606"},{"id":"cf35ec1cc6c8a262"}]}],"out":[{"x":560,"y":80,"wires":[{"id":"5ab77eb286d26d9a","port":0}]},{"x":560,"y":140,"wires":[{"id":"9cce0071663acf02","port":0}]},{"x":560,"y":200,"wires":[{"id":"b076cebd74637606","port":0}]},{"x":560,"y":260,"wires":[{"id":"cf35ec1cc6c8a262","port":0}]}],"env":[],"meta":{},"color":"#DDAA99","status":{"x":480,"y":160,"wires":[{"id":"5ab77eb286d26d9a","port":0},{"id":"9cce0071663acf02","port":0},{"id":"b076cebd74637606","port":0},{"id":"cf35ec1cc6c8a262","port":0}]}},{"id":"5ab77eb286d26d9a","type":"api-current-state","z":"7809448eb5724189","name":"If Overdag Verlichting","server":"8a29b7d6.5696f8","version":2,"outputs":2,"halt_if":"on","halt_if_type":"str","halt_if_compare":"is","entity_id":"input_boolean.woonkamer_fel","state_type":"str","blockInputOverrides":false,"outputProperties":[{"property":"verlichting","propertyType":"msg","value":"fel","valueType":"str"},{"property":"data","propertyType":"msg","value":"","valueType":"entity"},{"property":"payload","propertyType":"msg","value":"fel","valueType":"str"}],"x":280,"y":80,"wires":[[],[]]},{"id":"9cce0071663acf02","type":"api-current-state","z":"7809448eb5724189","name":"If Avond Verlichting","server":"8a29b7d6.5696f8","version":2,"outputs":2,"halt_if":"on","halt_if_type":"str","halt_if_compare":"is","entity_id":"input_boolean.woonkamer_aan","state_type":"str","blockInputOverrides":false,"outputProperties":[{"property":"verlichting","propertyType":"msg","value":"aan","valueType":"str"},{"property":"data","propertyType":"msg","value":"","valueType":"entity"},{"property":"payload","propertyType":"msg","value":"aan","valueType":"str"}],"x":270,"y":140,"wires":[[],[]]},{"id":"b076cebd74637606","type":"api-current-state","z":"7809448eb5724189","name":"If Nacht Verlichting","server":"8a29b7d6.5696f8","version":2,"outputs":2,"halt_if":"on","halt_if_type":"str","halt_if_compare":"is","entity_id":"input_boolean.woonkamer_dim","state_type":"str","blockInputOverrides":false,"outputProperties":[ [... cut ...] 6.5696f8","version":3,"debugenabled":false,"service_domain":"scene","service":"turn_on","entityId":"scene.woonkamer_tv_meubel_{{verlichting}}","data":"","dataType":"jsonata","mergecontext":"","mustacheAltTags":false,"outputProperties":[],"queue":"none","x":1130,"y":300,"wires":[[]]},{"id":"8a29b7d6.5696f8","type":"server","name":"Home Assistant","version":1,"legacy":false,"addon":true,"rejectUnauthorizedCerts":true,"ha_boolean":"y|yes|true|on|home|open","connectionDelay":true,"cacheJson":true}]`

ThomDietrich, 2021-10-05:
That's impressive!

ThomDietrich, 2021-10-08:
@Post404 if you want to give it a try: https://github.com/fabianoriccardi/melody-player/blob/master/examples/1_simple_play/1_simple_play.ino

Post404, 2021-10-08:
Thanks for your explanation. That make a lot of sense.
The main goal was to have a clear indication off when the Senseo is turned on by an automation. That goals is achieved, everything from now on is just polish. Also most of the time the tune plays good (enough) and if not, then it still achieved the notification.
At moments like this I wish I was better at coding. I'll still have a look at what you send after and see what I can do.

I really had fun with making the tunes so far. At this point I'm so invested in this project and the level off overkill features is already so high that this doesn't stand out that much . And it has a high WAF, I don't even drink coffee myself.