ThomDietrich/SenseoWifi #42: 🙋📈 How many active users do we have?

ThomDietrich, 2021-02-19:
Hey all,
for a long time, I was wondering how many people did get inspired by this project and built their own SenseoWifi. This little informal survey is purely voluntary and I'm looking forward to see where we get :)

Please click whichever applies to you:

* 🚀 - My SenseoWifi is up and running
* 🎉 - I did my own additional modifications
* 😕 - I was intrigued but failed to complete the rebuild

If you have a special story to tell, leave a comment.
Thanks all

m4rcu5, 2021-02-19:
:roll_eyes:, not because of the project, but because the plastic parts of my Senseo crumbled when taking it apart. I will take this on when the next machine crosses my path!

ThomDietrich, 2021-02-19:
@m4rcu5 really! I took apart three-four of them and multiple times during development. Maybe yours was old and the parts hard and brittle? If you ever end up buying a new one contact me and I'll send one of the left-over custom PCBs your way!

Post404, 2021-02-25:
Everything in this project has been new for me, ordering the parts, ordering a custom PCB, soldering, dealing with firmware and implementing it with MQTT. But after working on it for a long time and with some help I got it up and running yesterday and i'm very happy with it. I am going to add the cup-detector soon. The joke is that I don't even drink coffee, but my girlfriend is really happy with it. My guess is that it will mostly will be used to turn it on in advance when waking up, I don't think that preparing everything beforehand and then making a cup is going to happen a lot, but this will definitely happen when there are visitors.

It is fun to have these functions now, but the biggest part for me was to learn all the new things and get closer to creating something myself later. For example a motorized roller blind.

muchtkutte, 2021-03-14:
Hallo,

mein ausländisch ist leider sehr miserabel.

Ich habe mir die Platine vor einem halben Jahr zusammen gelötet und in meiner Senseo verbaut. Bei der Gelegenheit habe ich gleich einen Tassenzähler hinzu gefügt welcher aktuell bei 1104 steht. Ganz schön viel Kaffee getrunken. :)

Leider geht bei mir die "Automatik" nicht mehr da mein Tassensensor nichts mehr macht, eventuell ist es neben dem Boiler zu heiß. Ich werde dieser Tage das Gerät nochmals zerlegen und diesen erneuern. Bei der Gelegenheit möchte ich gleich einen Deckelsensor (mir Reedkontakt) verbauen, da ich festgestellt habe das die Senseo auch Kaffee kocht wenn der Pad-Deckel geöffnet ist. Das ist natürlich nicht so schön und mein Bastelwahn sagt das muss geändert werden.

Bei der Gelegenheit mal eine Frage - kann ich den Tassensensor irgendwie über ein MQTT Kommando deaktivieren? Ich habe die Maschine mit meinem Wecker gekoppelt und momentan ist das nicht so schön das ich früh's noch auf die Tassentaste drücken muß. Wasser erhitzen geht zum Glück noch. ;)

ThomDietrich, 2021-03-14:
Hey @muchtkutte,
die Amtssprache auf GitHub ist english aber ich kann dir gern auf Deutsch antworten ;)
Tassenzähler: Toll! Habe ich selbst auch, allerdings auf der anderen Seite von MQTT, also in der Heimautomatisierung bzw in miener InfluxDB.
Mit dem Tassensensor hatte ich nie Probleme. Vielleicht ist deiner willkürlich Opfer eines Fehlers geworden. Ersetze ihn doch einfach mal. Per MQTT umgehen geht leider nicht. Das habe ich nie als Funktion vorgesehen. Solltest du dafür den Code entsprechend erweitern kannst du aber gern einen Pull Request beitragen!
Die Idee mit dem Deckelsensor finde ich SUPER! Wir haben es damals nicht übetreiben wollen und wollten den mechanischen Aufwand gering halten. Wenn du deine Lösung implementiert hast stell doch bitte Aufbau, Code und Fotos uä hier bereit. Vielleicht nehme ich das auch in unseren Aufbauten auf :)

Freue mich über eine ANtwort, um den Thread hier sauber zu halten aber bevorzugt als eingenes Ticket :) Danke!

Vergiss nicht oben die 🚀 und 🎉 als Rekationen hinzuzufügen

Pierre-33, 2021-03-24:
I actually found about this project today, and I totally want to try that :)

I'm new to the electronic part of the project, so it's probably going to be quite a challenge. The end game for me would be to interface it with home assistant, probably through ESPHome. But I might play with your firmware instead, we'll see once I get here :)

I do have questions, though :) It would seem you have a 1.8 version of your pcb but the gerber export doesn't seem to exist for the 1.8. Am I missing something?
For the component parts, I found a PC817X3NSZ9F but it's not clear to me if it's equivalent to the  PC817 you prescribe or if it's a totally different model.

Thanks for this project! :)

Hakzel, 2022-03-27:
I stumbled across this amazing project and I wanted to automate my Senseo as well, even though I am not the coffee drinker in the house ;-)

Do you have a good source for the PCBs? If I try to order them online a quantity of 1 results in prices of 50€ which is a little bit much too handle. I was looking online to find a kit with all the parts but unfortunately without success. 

I am also from Germany so any hint into the right direction is highly appreciated :-)

ThomDietrich, 2022-03-28:
Hey @Hakzel, it becomes a lot cheaper if you e.g. order 10 via PCBway or similar. I should still have a few lying around (not the newest version but good enough to succeed). Send me an email and I can mail one to you.

Hakzel, 2022-03-28:
Will do, thanks a lot!

Hakzel, 2022-03-30:
@ThomDietrich thanks for the offer. I already sent you a mail but didn't get any response so far. Any other way to contact you? :-)

ThomDietrich, 2022-03-30:
@Hakzel of course! Let me answer you now ;)

Hakzel, 2022-04-14:
Happy to announce that the SenseoWifi works with the Senseo Quadrante! 
Successfully  modded my machine last weekend and it's brewing like a champ now

TonKro, 2023-04-21:
Very interesting project. Can I buy a BCB from someone?

ThomDietrich, 2023-04-21:
@TonKro Baarschot in the Netherlands? Maybe ask here https://github.com/ThomDietrich/SenseoWifi/issues/50

aguiller31, 2025-08-24:
Hey, I just finished modding my Senseo ! Thank you from your amazing work ! Having a full project with PCB, code and a perfect doc with pictures is a blast !! When of my favourite project, nothing to add, it's just perfect !! Thanks again !