ThomDietrich/SenseoWifi #43: It's probably quite the challenge

ThomDietrich, 2021-03-24:
I actually found about this project today, and I totally want to try that :)

I'm new to the electronic part of the project, so it's probably going to be quite a challenge. The end game for me would be to interface it with home assistant, probably through ESPHome. But I might play with your firmware instead, we'll see once I get here :)

I do have questions, though :) It would seem you have a 1.8 version of your pcb but the gerber export doesn't seem to exist for the 1.8. Am I missing something?
For the component parts, I found a PC817X3NSZ9F but it's not clear to me if it's equivalent to the  PC817 you prescribe or if it's a totally different model.

Thanks for this project! :)

_Originally posted by @Pierre-33 in https://github.com/ThomDietrich/SenseoWifi/issues/42#issuecomment-806024204_

ThomDietrich, 2021-03-24:
Hey @Pierre-33

The firmware can already be used with Home Assistant. Find my configuration here: https://github.com/ThomDietrich/SenseoWifi/issues/41
I was also thinking about eventually moving to another framework to be free in which MQTT convention the project communicates with. ESPHome is imho not suited for the project. I'd rather have a lightweight framework tailored to the project. We need a web frontend to configure Wifi and MQTT settings, the rest goes through standard mqtt library. I was looking into something like this: https://github.com/rjwats/esp8266-react

> It would seem you have a 1.8 version of your pcb but the gerber export doesn't seem to exist for the 1.8. Am I missing something?

We did update the PCB over time, however I never had 1.8 produced and hence did never export what I can not verify. You can simply load the project in Fritzing and export to gerber yourself.

> For the component parts, I found a PC817X3NSZ9F but it's not clear to me if it's equivalent to the PC817

As it starts with PC817, yes. The remainder is just details this project does not scratch. Just be sure to get the right form factor ;)

Pierre-33, 2021-03-24:
> We need a web frontend to configure Wifi and MQTT settings
I though the whole point of ESPHome was to take care of that. The wifi configuration seems to be written within the initial firmware and the communication between Home Assistant and the ESP handled without the need of an MQTT server. And as a bonus it seems to support OTA update out of the box.

> We did update the PCB over time, however I never had 1.8 produced and hence did never export what I can not verify. You can simply load the project in Fritzing and export to gerber yourself.
Well, if 1.8 has never been tested, I rather stick with the reliable one :)

> As it starts with PC817, yes. The remainder is just details this project does not scratch. Just be sure to get the right form factor ;)
It does seem to match. I'll probably place my order tomorrow :)

m4rcu5, 2021-03-24:
It is funny that you mention the firmware. I actually stumbled onto this project because I was researching the https://github.com/homieiot/homie-esp8266 project, which is the framework used by this project as well. It served as a really good example for what I was building.
It's a shame the development of the homie-esp8266 project seems to be stalled. I loved the simplicity and MQTT standard it had. I build a module for my air purifier with it :smile: 

I felt that ESPHome was a bit to rigid of a framework to build custom devices, which are more then turn thing on/off. But I shall take another look as well.

@ThomDietrich that esp8266-react project looks rather fancy, Though the "react" part scares me away, as front-ends are really not my forte.

ThomDietrich, 2021-03-24:
> Well, if 1.8 has never been tested, I rather stick with the reliable one :)

No don't take that the wrong way. It's the latest version and it would be a shame to work with what is already outdated. I simply did not upload the gerber files. If you are not sure how to do it I can get Fritzing and do it. It's the same effort for me as it is for you :)

> I though the whole point of ESPHome was to take care of that. 

ESPHome is great and has its place in the world where you want to connect a few sensors or actors to the outside world. The SenseoWifi project has a complex'ish logic around the state of your Senseo. It does not expose sensors but rather the machine as a functional system. It's simply not the best way to approach this.

The currently used framework homie-esp8266 does it right. It offers a nice web frontend to configure Wifi, MQTT, and further custom settings (like whether you have a cup detector or not) and then facilitates MQTT communication according to the Homie convention.

@m4rcu5 most of the existing logic can stay as it is. I just want to replace Homie by the standard MQTT library and the Home Assistant format, and replace the Homie web frontend by something just as useful. If you know any project that offers that, let me know!

Pierre-33, 2021-03-25:
I never used Fritzing, if it's as simple as clicking the export button, I can probably purchase the software and do it myself, if it's more complicated than that with a lot of export option, I might get lost :)

Ok, I understand the difference between your firmware and ESPHome, my approach was to write all the "intelligence" within homeassistant (using AppDaemon which is so great btw), but I can see the value of your approach. I'll see what I will do once I get there :)

ThomDietrich, 2021-03-25:
Right! They removed the option to download for free! Okay give me 1-2 days and I can export to gerber. How are you going to produce the PCB? There are a few online providers and they massively differ in cost...

Intelligence in home assistant is imho not a good idea, as you need to observe the quickly changing LED. I get your general idea but let me humbly say: As there is already a fully working firmware I wouldn't know why you would go the more efficient way :)

Pierre-33, 2021-03-25:
Thanks, that would be great :) I was thinking about going with aisler.net. They are a bit more expensive than the Chinese companies, but they are in Europe and you order some of the components directly from them. The biggest downside is that you need to order at minium 3 PCBs, but giving my soldering skill, it's probably not such a bad idea to have some spares :)

I'll probably follow your advice regarding the firmware.

Yaad182, 2021-07-11:
I'm thinking of doing this project too, except I have 0 experience with soldering and PCB's, so this'll be interesting... Any tips on where I can buy all these parts including the custom PCB? I live in the Netherlands :)

ThomDietrich, 2021-10-04:
Hey all,
probably too late by now but I have just uploaded the v1.8 gerber export. Hope it's useful for some.

@Yaad182 there are many cheap or fast or convenient options. You can order the PCBs via https://www.pcbway.com and the components should be available on ebay. Can't really give the one best answer here :) Good luck!

closed 2021-10-04 by ThomDietrich