ThomDietrich/SenseoWifi #47: OTA update

Post404, 2021-11-09:
I've tried a few times to update OTA, but I never got it to work. What am I missing here?
In Visual Studio Code I changed the ip, username, password and the senseo name in de platformio.ini file.
added "upload_port = 192.168.xx.xx"
added "build_dir = mybuildfiles"
changed the file location from "$BUILD_DIR/firmware.bin" to ".pio/build/senseo-wifi/firmware.bin"
Uploading goes a long way and seems to work for the most part but ends with:
```
Traceback (most recent call last):
  File "D:\Projecten\Home Assistant\Wifi Senseo\SenseoWifi-master 9 nov\SenseoWifi-master\ota_updater_202011.py", line 4, in <module>
    import paho.mqtt.client as mqtt
ModuleNotFoundError: No module named 'paho'
*** [upload] Error 1
```

ThomDietrich, 2021-11-09:
Hey!
Uploading directly through PlatformIO doesn't work. That's out of the box only possible via USB, not OTA.

First of all, try from a normal command line. The command is given here: https://github.com/ThomDietrich/SenseoWifi#ota-updates (you can use the command line inside VSCode, if the error from above pops up you need to execute once `pip3 install paho-mqtt`.

After you got the command working I did actually implement the change to make all of this work right out of PlatformIO two days ago: https://github.com/ThomDietrich/SenseoWifi/blob/b9e0cf8c536b316d8d3fe9397ac7d7e5beb8d335/platformio.ini#L20-L21 (you need to change the details to your setup)

Post404, 2021-11-09:
Everytime I updated I tried to get OTA working and then not get it to work and shamefully dissasemble it again.

Allright I'm trying the command through terminal now. What I did wrong is that I used the IP of the Senseo and not of Home Assistant and I had to change python3 to py. So that is fixed now.
When using the command in VScode it is saying that the device is already up to date. Perfect!
So from now on I can just download updated files, replace them and use the command again to update.
Without having to dissasemble the thing over and over.

Is there an easy way to pull the latest files?

I did also updated the upload_command in platformio.ini.
When using the task button "Upload" or "Upload Filesystem Image OTA" it's still saying it is missing paho.
I allready installed that (version 1.5.1).

ThomDietrich, 2021-11-09:
The devil is in the details, can't say for sure but hope you get it working.

The important thing, upload via command line seems to be working, so that's amazing.

> Is there an easy way to pull the changed files?

Yes generally you should go with git. if you have no experience with it look at any online tutorial, it has some kinks to know about but is super useful. In short: you can always quickly `git pull` the latest changes (VSCode has support for that), then press "Build", then upload through command line. Voila!

As I said, please share either a PR or simply your thoughts about what needs to be improved in the README and I can take care of it.

Post404, 2021-11-14:
Thanks for the tip, I will look into this.
The only thing that might clear things up for beginners like me is that the command is easiest used in VScode.
And that it works by connecting to de MQTT broker. I was thinking it was directly somehow.

For now I'm trying to update with the command in VScode and it is saying it is already up to date with md5 checksum and then some long string of numbers and letters.
It's on 1.7.3, is this the latest one?

ThomDietrich, 2021-11-14:
Nope :)

https://github.com/ThomDietrich/SenseoWifi/blob/129a7b1b5c5ec63c72e101dc2d3d28da20b56185/src/SenseoWifi.cpp#L336

I might add a few sentences around the command inside the README

Post404, 2021-11-14:
Thanks! I got it working, I didn't do anything different, but I guess i'm a wizard.

closed 2021-12-23 by ThomDietrich