ThomDietrich/SenseoWifi #49: Entities not available in home assistant

bjornmorsman, 2021-11-18:
Hello Sir,

yesterday i updated my senseo with the latest firmware, now i have the following problem.

when i boot my senseo it works very good, but after +- 5 minutes al my senseo entities become unavailable in home assistant. The mqtt topics wont get updated anymore.

Is it just me or does someone else has the same problem?

sorry for my bad english🤭

ThomDietrich, 2021-11-18:
Hello,
no worries, we will sort this out.

Did you use this configuration? https://github.com/ThomDietrich/SenseoWifi#home-assistant-configuration-example

> The mqtt topics wont get updated anymore.

That shouldn't make the machine unavailable but the "available" topic is important here. Please check your MQTT available topic (`homie/.../$state`)

bjornmorsman, 2021-11-18:
Hello Thom,

yes i used your configuration for home assistant, it has worked for the past 8 weeks so i think the configuration is correct.

I just went from firmware 1.7.8 (the last one) to 1.7.3 (older one) first time downgrade didnt work because the ota updater doesnt see te senseo as online (unavailable in home assistant) after a reboot the ota update works again.

now i am at 1.7.3 and everything works correct, the senseo doesnt go offline (unavailable in home assistant) am i missing some steps with the ota to 1.7.8 maybe?

thanks in advance

ThomDietrich, 2021-11-18:
OTA is totally unchanged since the very first version of this project.

I did a lot of changes in the last couple of weeks, so you definitely should check out the latest config from the readme again!

See all changes here: https://github.com/ThomDietrich/SenseoWifi/commits/master

There shouldn't be any reason to use 1.7.3 vs 1.7.8. In fact the only two changes are the last two commits which handle edge cases and didn't change anything to how everything works.

ThomDietrich, 2021-11-18:
In case of doubt you can always connect via USB and do a fresh flash, which also deletes your configuration. This way you start without any remaining garbage in the memory and it must work...

ThomDietrich, 2021-11-18:
Did you disable either the buzzer or the cup detector?

bjornmorsman, 2021-11-26:
Hello Thom,

thanx for your advice, the problem was my wemos-d1-mini. it keeps disconnecting from the wifi, after a clean reflash the problem stayed so i ordered a wemos-d1-pro and that solved my problem.

thanks for you help and you can close the issue!
Keep up the good work

closed 2021-11-26 by bjornmorsman