ThomDietrich/SenseoWifi #50: List of hardware

mull0r, 2022-04-07:
Maybe (probably) I am an idiot, but where can I get my hands on a compatible PCB? I am not unfamiliar with flashing and soldering, however I am unfamiliar with custom PCBS. 

Is it an idea to provide the list of requirements and include one or more examples where to get the hardware? 

I am located in The Netherlands by the way.

bjornmorsman, 2022-04-07:
Ik heb nog 4 v1.7 pcb’s liggen, dus als je wilt kun je deze wel overnemen.

ThomDietrich, 2022-04-07:
Thanks @bjornmorsman,

generally you can just order the PCB to be produced. The gerber files you need to do so can be found here: https://github.com/ThomDietrich/SenseoWifi/tree/master/SenseoWifi-PCB

A cheap international option is https://www.pcbway.com but there are more than enough local option, if you want to suport those.

In additon to the PCB you need the electrical components to be soldered on, the list is in the README. Let us know if you need more help.

closed 2022-07-12 by ThomDietrich

timvanhulten, 2022-07-25:
> 

Zou ik er twee van je kunnen overnemen?

bjornmorsman, 2022-07-25:
Hoi Tim,

10tje exclusief verzendkosten? Kun je je daarin vinden?

Groet, Bjorn

timvanhulten, 2022-07-25:
Ja, is prima. Kan gewoon met briefpost neem ik aan. Kunt het verzenden naar Valeriusstraat 4A, 5151LP te Drunen. Kunt me toevoegen op whatsapp voor een tikkie. Via 0622522140.

Bvd

Outlook voor Android<https://aka.ms/AAb9ysg> downloaden

TonKro, 2023-04-21:
Heb je er nog liggen mogelijk? Ik heb er maar 1 of 2 nodig.

bjornmorsman, 2023-04-21:
Ik zal zo eens in mijn hobby la kijken, heb er zeker nog 1. De 2de moet ik even nakijken. Wil je de rest van de componenten er ook bij?

mvg,

Bjorn

TonKro, 2023-04-21:
Ik hoor graag. Componten is zeker zo makkelijk maar kan die eventueel ook bestellen.

TonKro, 2023-04-27:
Ben je er nog aan toegekomen?

timvanhulten, 2023-04-27:
Ik heb de print toevallig vorige week afgemaakt. Mistte de 1.5k ohm zekering, die inmiddels is toegevoegd. Volgende week bouw ik hem denk ik in. Mits ik niet weer een of ander project tussendoor oppak.

Verzonden vanaf Outlook voor Android<https://aka.ms/AAb9ysg>

TonKro, 2023-04-27:
Als ik je goed begrijp heb je geen (bestuukte) print meer over?

timvanhulten, 2023-04-28:
Heb er nog 1 over

Verzonden vanaf Outlook voor Android<https://aka.ms/AAb9ysg>

TonKro, 2023-04-28:
Die wil ik graag overnemen van je. Laat maar weten hoe we dit kunnen regelen

ThomDietrich, 2024-11-01:
Hey Konrad, I still have a bunch. I can send you one in a letter. Need to check but I think it is version 1.6 of the PCB, which is still already good enough. The later versions were just for perfection

qwibbert, 2025-04-05:
I'm from Belgium and I'm looking for the PCB board. I thought it would be easier and cheaper to just buy one off of someone who got multiple copies already printed.  Anyone who has a spare board laying around?