ThomDietrich/SenseoWifi #53: New MQTT configuration format

Post404, 2022-06-02:
Hey Thom,
They justed introduced a new MQTT configuration format.
There is a change needed in the HA configuration for the part with the switch, binary sensors, sensors and the template sensor.

This is the configuration i'm using right now, were I updated everything. I just didn't get the uptime template sensor to work again:

mqtt:
  switch:
    - unique_id: uniqueid__senseowifi_power
      name: senseowifi_power
      icon: mdi:coffee-maker
      #
      state_topic: "homie/senseo/machine/power"
      state_on: "true"
      state_off: "false"
      #
      command_topic: "homie/senseo/machine/power/set"
      payload_on: "true"
      payload_off: "false"
      #
      availability_topic: "homie/senseo/$state"
      payload_available: "ready"
      payload_not_available: "lost"

  binary_sensor:
    - name: senseowifi_out_of_water
      unique_id: uniqueid__senseowifi_out_of_water
      icon: mdi:water-off-outline
      #
      state_topic: "homie/senseo/machine/outOfWater"
      payload_on: "true"
      payload_off: "false"
      device_class: problem
      #
      availability_topic: "homie/senseo/$state"
      payload_available: "ready"
      payload_not_available: "lost"

    - name: senseowifi_cup_available
      unique_id: uniqueid__senseowifi_cup_available
      icon: mdi:coffee-outline
      #
      state_topic: "homie/senseo/machine/cupAvailable"
      payload_on: "true"
      payload_off: "false"
      #
      availability_topic: "homie/senseo/$state"
      payload_available: "ready"
      payload_not_available: "lost"

    - name: senseowifi_cup_full
      unique_id: uniqueid__senseowifi_cup_full
      icon: mdi:coffee
      #
      state_topic: "homie/senseo/machine/cupFull"
      payload_on: "true"
      payload_off: "false"
      #
      availability_topic: "homie/senseo/$state"
      payload_available: "ready"
      payload_not_available: "lost"

  sensor:
    - name: senseowifi_brewed_size
      unique_id: uniqueid__senseowifi_brewed_size
      icon: mdi:coffee-maker
      #
      state_topic: "homie/senseo/machine/brewedSize"
      #
      availability_topic: "homie/senseo/$state"
      payload_available: "ready"
      payload_not_available: "lost"

    - name: senseowifi_opstate
      unique_id: uniqueid__senseowifi_opstate
      icon: mdi:state-machine
      #
      state_topic: "homie/senseo/machine/opState"
      #
      availability_topic: "homie/senseo/$state"
      payload_available: "ready"
      payload_not_available: "lost"

    
    - name: senseowifi_debug
      unique_id: uniqueid__senseowifi_debug
      icon: mdi:comment-text-multiple-outline
      #
      state_topic: "homie/senseo/machine/debug"
      #
      availability_topic: "homie/senseo/$state"
      payload_available: "ready"
      payload_not_available: "lost"

    - name: senseowifi_rssi
      unique_id: uniqueid__senseowifi_rssi
      icon: mdi:signal-cellular-2
      #
      state_topic: "homie/senseo/$stats/signal"
      unit_of_measurement: "%"
      #
      availability_topic: "homie/senseo/$state"
      payload_available: "ready"
      payload_not_available: "lost"

    - name: senseowifi_uptime
      unique_id: uniqueid__senseowifi_uptime
      icon: mdi:av-timer
      #
      state_topic: "homie/senseo/$stats/uptime"
      unit_of_measurement: "s"
      #
      availability_topic: "homie/senseo/$state"
      payload_available: "ready"
      payload_not_available: "lost"

################################################################################

template:
  - sensor:
      senseowifi_uptime_since:
        unique_id: uniqueid__senseowifi_uptime_since
        friendly_name: "Senseo Laatste Herstart"
        value_template: >
          {% if is_state("sensor.senseowifi_uptime", "unavailable") %}
            {{ states.sensor.senseowifi_uptime.state }}
          {% else %}
              {{ ((as_timestamp(now() - timedelta(seconds=(states.sensor.senseowifi_uptime.state | int))) / 300) | round() * 300) | timestamp_local() }}
          {% endif %}
        icon_template: mdi:calendar-clock
        device_class: timestamp

ThomDietrich, 2022-06-02:
Amazing! Thanks.
I saw the change in the release notes and wanted to adapt my setup soon'ish. Thanks :)

In the meantime I've changed a few minor details as well. Let me try to upload my version of the changes before my vacation starting this weekend :)

ThomDietrich, 2022-06-02:
On a side note: I don't know the reasoning for this change but don't like it. Before each individual entity was clearly defined as an mqtt entity. Now it's quite ambiguous if you just look at, e.g.:

[image]

Post404, 2022-06-02:
For a package like this I understand you don't like it, but for the most use cases this is a separate file that is included as "mqtt: !include mqtt.yaml". In that case you know you are in the mqtt file. I think this change is done for consistancy, it's build up the same as all the other things now. which in turn makes it easier to understand for new users.

ThomDietrich, 2022-06-02:
Yeah might be the case

I'm going on a leap here: Everyone who doesn't use a package based approach to Home Assistant configuration is wrong and people should rethink their structures. All other approaches do not scale to a meaningful setup and every snippet or tutorial you find on the web provides config elements with the assumption of packages.
In fact, I wonder if an ha-architecture discussion should be started to introduce a package structure as the default in the new install config example. I should put that on my list of todos.

ThomDietrich, 2022-06-02:
From https://github.com/home-assistant/core/pull/71676

> The background for this PR is that the MQTT integration itself is setup via a config entry, with entities setup via discovery.
>MQTT entities can also be setup through configuration.yaml though, and those entities are not aware of the MQTT config entry and thus are not removed if the MQTT config entry is disabled or removed.

Apprarently this was driven by a technical limitation :)

ThomDietrich, 2022-06-02:
Btw do you use the brew cycle automation with the repeated beeping as I implemented it? I have occasional issues with it (beeping continues) and can't quite figure out the cause.

Post404, 2022-06-13:
Thanks for the info!
I turned off that automation, I was thinking it was me doing something wrong.
The only thing I can think of is that it is just repeating the beeping without checking the condition.
I can't test it right now though.

ThomDietrich, 2022-07-12:
Hey,

sorry took me so long but I've now also updated my config. It also includes a few other bits.
https://github.com/ThomDietrich/SenseoWifi/blob/master/home-assistant/senseo_wifi.yaml

I did also fix the issue with the automatic brewing cycle. It was rather silly. You should have a look as it is super useful. I use it combination with an Ikea push button, which sets `input_boolean.senseowifi_brew_double_automated` to ON.