ThomDietrich/SenseoWifi #56: Second try :)

Pierre-33, 2023-02-03:
Hey! I'm back :) After almost 2 years, I wan't to give it another try. Printing the pcb online was a bit daunting so I want to give it a try with a protoboard. I do have a few questions though (forgive me for the dumb one)

- what are the resistor for? Can't you just connect the side of the octocoupler that emit light directly to an out pin of the ESP? 
- I understand from the schematic what side of the octocoupler should be connected toward the esp and which should be toward the senseo, but I have no idea how to know what is what from the actual piece of hardware. Is there any convention to know which part emit light and which part receive them?
- I ordered a HLK-PM01, but it seems tricky to use. I dig a bit on the internet and it seems to involve using fuse (https://randomnerdtutorials.com/esp8266-hi-link-hlk-pm03/) and I'm not very confortable soldering a custom pcb for 220v (and it seems the pin of the HLK doesn't want to fit into any of my protoboard anyway). Can you update the reference of the step down converter you use? (the link on ebay is broken)

The last things is more of a suggestion. Have you considered to hack directly the buttons into the esp to add nice behavior like 
- Pressing the cup button could trigger the full brewing cycle. (turn on, wait for warm up, brewing, turn off)
- A long press of the cup button could turn on an mqtt switch that would be used by home assistant (or your favorite automation) to program a coffee for the next morning
- A long press of the main button could reset your firmware (no need for an extra button)
- Requesting a coffee with no cup present could be rejected and trigger a sound from the buzzer

ThomDietrich, 2023-02-03:
Hey! 
I will respond to everything later today. Just a quick question though: Where do you live? I could send a letter with one of my spare PCBs your way.

Pierre-33, 2023-02-03:
Thanks, that would be nice :) I live in France. 
But depending on your answer on the second part of my message, I might want to try the improvement I propose. I could contribute to the c++ implementation with a little help on the electronic part (mostly where to put resistor)

Pierre-33, 2023-02-05:
A little update on my side. I did some reading this week-end and now I know how to read the octocoupler pinout and manage to use one for another simpler project. I did find out as well that without the resistor the octocoupler burn, so the resistor must be useful :) I just don't know how to calculate the needed value yet, so I just use the same as you.

ThomDietrich, 2023-02-06:
Hey!

Jup that pretty much answers it. Looking at the [datasheet](https://global.sharp/products/device/lineup/data/pdf/datasheet/PC817XxNSZ1B_e.pdf) we learn that pin 1 is marked by a shiny dot on the top of the chip ("Anode mark"). On my PCB it is marked by a rectangular pad. The resistor follows the standard rules for an in-series resistor for an LED. Seems like I went with a bigger resistor, which in this case does not hurt and is often good practice.
Try yourself (3.3V, 1.2V. 5..20mA): https://www.digikey.de/en/resources/conversion-calculators/conversion-calculator-led-series-resistor

Am I correct to assume that you are on a beginner level in electrical engineering?

> I'm not very comfortable soldering a custom PCB for 220v

And you definitely shouldn't! Please buy a ready to be used component as I did, or permanently power the D1 Mini with a separate USB power supply.
This is my solution: https://raw.githubusercontent.com/ThomDietrich/SenseoWifi/master/images/DSC09646.jpg (no own soldering on the 230V side) https://www.ebay.de/itm/232569563115?mkcid=16&mkevt=1&mkrid=707-127634-2357-0&ssspo=vdx7hCYXSwa&sssrc=2047675&ssuid=LwPuJZgXSg2&widget_ver=artemis&media=COPY

> Have you considered to hack the buttons directly into the ESP to add nice behavior like

I did consider this a couple of times. We didn't feel the additional components and effort were justified. It's certainly possible to disconnect the buttons from the Senseo and directly to the ESP GPIO pins. One blocker today: The D1 Mini does not have enough GPIOs free to connect three buttons.
In the beginning we didn't want to break default behavior of the machine, later we realized it wasn't needed. I still mostly handle it by hand, eventually we use Alexa or an Ikea Zigbee button and Home Assistant to start a complete brewing cycle.
I find your suggestion intriguing and would consider it for a new project, here and today I don't see the need. 

Happy to send over a PCB if you carry the postal service fee.

Pierre-33, 2023-02-06:
> Am I correct to assume that you are on a beginner level in electrical engineering?

Yes, you are totally right :) I purchased a book recently and having a lot of fun! Hopefully, I should be soon able to calculate the right resistor value myself :)

> This is my solution: https://raw.githubusercontent.com/ThomDietrich/SenseoWifi/master/images/DSC09646.jpg (no own soldering on the 230V side) https://www.ebay.de/itm/232569563115?mkcid=16&mkevt=1&mkrid=707-127634-2357-0&ssspo=vdx7hCYXSwa&sssrc=2047675&ssuid=LwPuJZgXSg2&widget_ver=artemis&media=COPY

I assumed the red wire on your picture are the 220v one, right? So you still have to solder your own cable (obviously bigger section) to the step-down PCB? Is that safe? I feel like the part where the red wire meet the step-down PCB is totally unprotected. 
I find this one which seems quite similar : https://www.amazon.fr/AC-DC-700mA-Power-Supply-Module/dp/B08KH82YK1/ref=sr_1_6?keywords=220v+to+5v&qid=1675692136&sprefix=220v+t%2Caps%2C199&sr=8-6 but I don't understand why there is 3 output pin. Vc++, GND - and a mysterious output. Does yours as that?

On that same topic, if I understood correctly it's safe to have the ESP connected to the computer with the USB cable AND the Senseo operating on 220v at the same time? (Of course the Senseo cover should be put back in place or extra caution should be taken not to touch anything inside while it's powered on).

> The D1 Mini does not have enough GPIOs free to connect three buttons.
Oh, I didn't think about that :-/ I didn't open my Senseo yet, how much space do we have inside? Do you think a full-size esp 8266 would fit in or did you pick the mini because of the tight space?

> Happy to send over a PCB if you carry the postal service fee.
I'll see how it goes with my test and I might ask you for one when I'm ready, thanks for offering.

ThomDietrich, 2023-02-06:
> I purchased a book recently and having a lot of fun! Hopefully, I should be soon able to calculate the right resistor value myself :)

This is great. Carry on! 🎉 

> I assumed the red wire on your picture are the 220v one, right?

As you can see, I used T-tap connectors on the wire side: https://www.amazon.fr/120-pi%C3%A8ces-Connecteurs-%C3%A9lectriques-tap/dp/B077YB123S However, you might be right, the red wires are soldered on the PCB side. In this case it is totally fine: The consumption of the additional electronics is minimal, so is the power over the AC input of the power supply. 

No need to be scared by these steps, but cautious. Keep the machine disconnected; install T-taps and ensure proper soldering results; feed 5V wires out of the machine for testing; ensure that the power supply is properly fixed and protected inside the Senseo (I suggest to hot glue to the floor). Only after everything is installed and sealed away start your first testing and measure that stable 5V are supplied.

As before: Do not connect AC power (Senseo and/or AD-DC power supply) and USB at the same time. This is first and foremost a safety precaution. The Senseo is not properly AC separated. Somewhere between the Senseo electronics, the AC-DC power supply and your laptop/PC you might encounter voltage/potential differences which can give you a scare, damage your ESP, or your Laptops USB port. 

> Vc++, GND - and a mysterious output

Not sure what that is but the component looks okay otherwise

> Oh, I didn't think about that :-/ I didn't open my Senseo yet, how much space do we have inside?

If you really wanted to make this happen, you could work with pin A0, which is not actually used in conjunction with the TCRT5000. This looks like a nice explanation: https://electrocredible.com/multiple-buttons-on-one-analog-pin-arduino-adc
The inside is not spacious but you would probably be able to add a bigger microcontroller. Beware that the ESP8266 only offers up to 11 GPIOs and this project here does only support the ESP8266.

Pierre-33, 2023-02-06:
> However, you might be right, the red wires are soldered on the PCB side. In this case it is totally fine: The consumption of the additional electronics is minimal, so is the power over the AC input of the power supply.

This is the part that didn't make sense to me, I didn't realize that since the power supply draws very little power, there was no danger on the power supply PCB

> If you really wanted to make this happen, you could work with pin A0, which is not actually used in conjunction with the TCRT5000. This looks like a nice explanation: https://electrocredible.com/multiple-buttons-on-one-analog-pin-arduino-adc

This is brilliant! I love that idea :) So you don't use the analog output of the TCRT5000 even if it's wired up in your diagram into A0 and use only the digital output wired to D6?

ThomDietrich, 2023-02-06:
Correct: https://github.com/ThomDietrich/SenseoWifi/search?q=cupDetectorAnalogPin
The TCRT5000 has a digital output telling us whether a cup was detected or not. The analog value does not provide any further benefit. I probably planned it in for testing and never removed the link. Maybe I will do that in the next PCB version (I intend to modify a "Senseo Select Eco" soon)

> I didn't realize that since the power supply draws very little power, there was no danger on the power supply PCB

The high voltage is still there and dangerous :) but the current flow over the solder points is small, hence you don't have to worry about heat damages

Pierre-33, 2023-02-19:
Just a quick message to tell do that I flashed the firmware yesterday and I'm pretty impressed with the setup process. I know it kind of come for free with Homie, but it add a nice professional touch to the whole things :)
For now it's all wires and breadboard on my side. I didn't cut open the Senseo yet (waiting for some parts). 

In the meantime, I'm working on automatic Home assistant discovery. I have managed to publish the proper message with MQTT explorer, I just need to find a way to make the getMqttClient()->publish() method works now :) (for some reason it doesn't seem to be connected when the MQTT_READY event fire).

Pierre-33, 2023-02-20:
Got it working, it wasn't a connection issue, it was just payload that was too big for async-mqtt-client (even if they were only 300 bytes). I had to bump async-mqtt-client to 0.9 to make it work. I hope it won't interfere with Homie usage of the lib, but so far so good.

ThomDietrich, 2023-02-20:
> I'm working on automatic Home assistant discovery.

Interesting.
So just fyi: I intend to hack the Senseo Select Eco with the same PCB (new version for modifications where needed), but I want to rely on ESPHome this time. Homie was cool back when I started this project but sadly did not catch on. ESPHome and the HA MQTT convention are what I want to rely on going forward. I am not promising anything, but just so you know. 

Keep me updated on your journey! ;)

Pierre-33, 2023-02-25:
I finally reach the interesting point where I open my Quadrante open. I can understand why it's easier to let the button working as they are and just solder on the back. I still wan't to give my idea a try though and I'd like to plug myself between the male and female connector you can see on the picture below. For that I need to know the name of those kind of connectors to see what I can find on internet :) If by any chance you happen to know it, it would be very helpful 

[image]

Pierre-33, 2023-02-25:
Me again :)
Go it working with a custom PCB in the middle and a bunch of Dupont cable, but I'm still interested in the name of the connector if you have it. I'm not sure the PCB will fit under the plastic cover.

This kind of work, but the behavior is a bit erratic, so I could use some clarification/help on a few points :
- The LED Observer seems not to trigger sometimes and it results in wrong states in HA. I don't have a clear repro yet. Is it supposed to work, both when you remote control the Senseo, and when you operate it manually? Even if you mix both modes of operation?
- Brew two coffee works, but brew one doesn't. Most likely some bad wiring on my side that I'll figure out, but the weird thing is that I don't get anything in the log. Should I get a message in the log if the MQTT part was working correctly?
- Is "Go through a complete brewing cycle automatically" managed only by a HA automation or is it managed directly on the ESP? If so how am I supposed to trigger it?
- Once, the Senseo started brewing automatically after being turned on. Is it supposed to retain previous calls to Brew x Cup?

Ps: on my breadboard I use a "regular" 8266, not a D1 mini, don't know if could have any impact.

ThomDietrich, 2023-03-09:
Hey,
nothing wrong with your Dupont cable solution :) Check this out: https://de.rs-online.com/web/p/leiterplatten-header/7121817

> So do you intend to write a "SenseoComponent" for EspHome

No not really. I would just expose digial inputs and outputs as entities to Home Assistant, which HA can then work with based on automation rules as we already do. I would need to investigate how to best implement the LED and Senseo state machines, maybe a custom component is needed for that.

> It sounds like a totally different project to me.... sounds a bit like a step back

It still uses the same PCB and the same logic, just with a different software framework underneath. With ESPHome the project would indeed be more focused at HA, however ESPHome can be connected to all other systems via MQTT. We are not excluding anyone, technically.
The advantage is the switch from the dead Homie project to the actively developed ESPHome project. Debugging, onboarding, and continued development should be simpler. I must add though: I do not know when I will be able to tend to this step.

> Does it make sense I create a pull request

Yes absolutely! Please do! I can test your PR with my working machine and feed back any issues I observe.

> The LED Observer seems not to trigger sometimes and it results in wrong states in HA. I don't have a clear repro yet. Is it supposed to work, both when you remote control the Senseo, and when you operate it manually? Even if you mix both modes of operation?

The LED times should always be accurate. If not, you are bound to experience issues. Enable `testIO` and verify the measurements.

> Is "Go through a complete brewing cycle automatically" managed only by a HA automation or is it managed directly on the ESP? If so how am I supposed to trigger it?

It was part of the machine code before but that wasn't dynamic enough. The new way is through the set of automations in Home Assistant, which work flawlessly. https://github.com/ThomDietrich/SenseoWifi/blob/master/home-assistant/senseo_wifi.yaml#L247
Triggered simply by switching `input_boolean.senseowifi_brew_double_automated`

> Once, the Senseo started brewing automatically after being turned on. Is it supposed to retain previous calls to Brew x Cup?

Didn't understand that.

> Is it possible to access the log via wifi?

There is a "debug" topic on MQTT.

> Ps: on my breadboard I use a "regular" 8266, not a D1 mini, don't know if could have any impact.

Doesn't. It just wouldn't fit on my PCB design.

Pierre-33, 2023-03-09:
A lot of progress has been made since my last post :)

I prototyped the voltage divider to read all the buttons with A0, it turns out to be more complicated than expected because the switch share their ground wire with the LED but I managed to make it work and I intend to put that part of the circuit in the little PCB on my picture which might make it compatible with your PCB. As a nice addition, I gain control over the Senseo Led which allow me to handle custom blinking patterns that actually match the esp state machine. Now brewing is 2s on and 200ms off. 

I find out that not all Senseo are born equal. Mine doesn't have the feature that turn-on, warm up brew and turn off when you press the cup button. Well... it was before I took control over the input button :) Now it's handled by the esp firmware!

Most of my weird behavior was caused by me sending command to senseo esp via MQTT while the Senseo wasn't plugged and then plugging it afterwards. It seems the FSM was totally desync after that. It shouldn't happen during normal operation and it is only possible because it's powered on USB for now.

So what next for me?

- Rewrite the FSM with the new buttons . I should be able to do something more robust as I don't have to guess from the LED state if I'm in brewing or heating
- Holding the power button will beep 3 times and then reset the machine
- Pressing cup button will trigger a full brewing cycle (almost done, I just have to find out why it doesn't turn off the machine)
- Pressing cup button without a cup will trigger an error beep instead of brewing (or maybe a specific led pattern)
- Holding cup button will raise a binary_sensor in Home Assistant that will in return trigger a coffee for the next morning when I woke up. I will probably use a very slow blinking pattern when a coffee is pending (like once a minute). 

I feel like doing something stupid like turn off the light of my kitchen when I press two buttons of my Senseo, just for the fun of it. But unfortunately the voltage divider doesn't give me different reading if I press more than one switch at the same time, if you have any tricks left in your bag concerning that, I'm interested :)

ThomDietrich, 2023-03-09:
Wow that's great progress! Congrats!!
I am happy with my old Senseo as it works today and won't apply thee changes. However, maybe I could with the new machine. :D

Voltage divider: It is possible to detect more than one button in parallel. The theory is that you can dimension your circuit to generate something like a binary encoding:

* Button 1 pressed: Value=1
* Button 2 pressed: Value=2
* Button 3 pressed: Value=4
* Button 4 pressed: Value=8

* Value=5 --> Buttons 1 and 3 pressed

The problem with this is the margin of error! You should open Excel and play around with some combinations of resistors to see how you might keep all relevant values reasonably apart from each other. Google for "resistor ladder binary" to learn more: https://www.best-microcontroller-projects.com/R-2R-ladder.html

Pierre-33, 2023-03-09:
Bitfield with resistor!! :) Once again it's exactly what I was looking for, thanks a lot!