ThomDietrich/SenseoWifi #63: GUI not visible

timvanhulten, 2023-08-05:
I succeeded in building and flashing the firmware. However the GUI is not visible. I cant enter the WiFi and MQTT data. I then tried to flash the config with the WiFi and MQTT credentials. This worked fine for connecting to WiFi. But connecting to my mqtt broker didn't work. 

Does anyone still get this to work?

ThomDietrich, 2023-08-07:
Hey there,
your problem is not quite clear. You need to follow these instructions until you reach the web interface described in step5:  https://github.com/ThomDietrich/SenseoWifi#firmware-upload 
If you can't reach it you did not succeed...

TonKro, 2023-08-11:
I guess you mean that when you connect to the WiFi access point that it returns on the web page:

Received captive portal request: UI bundle not loaded. See Configuration API usage: http://homieiot.github.io/homie-esp8266

I do see the same issue, did you find a solution?

ThomDietrich, 2023-08-11:
Under step 4 it says: "Upload File System image" -> This should provision the UI bundle. Please check

TonKro, 2023-08-11:
Can you help me with that step, I only have the firmware.bin to upload. Platform IO only could upload that. What do you mean with File System image?

TonKro, 2023-08-11:
Further checking with Upload File System Image, then get the bellow error:

Stub running...
Configuring flash size...
Compressed 1024000 bytes to 65410...
Writing at 0x00300000... (25 %)
Writing at 0x00304000... (50 %)
Writing at 0x00308000... (75 %)
Writing at 0x0030c000... (100 %)
Wrote 1024000 bytes (65410 compressed) at 0x00300000 in 6.9 seconds (effective 1187.2 kbit/s)...

A fatal error occurred: Timed out waiting for packet header
*** [uploadfs] Error 2

ThomDietrich, 2023-08-11:
That's a command in Platformio.
See here: https://www.donskytech.com/how-to-upload-filesystem-image-in-platformio-ide/ and pay attention to the fact that the project repository comes with a `data` folder as described in the article.